prusa3d/Prusa-Firmware-Buddy #4151: [BUG] Impossible to change diameter of nozzle

ClaGre, 2024-08-23:
### Printer model

XL multiple

### Firmware version

6.1.2

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

After upgrading from 6.0.4 > 6.1.2 I can't chose different nozzle diameter from the first and second head (I have two head configuration but I think 5 head have the same issue).
I've followed this guide: [https://help.prusa3d.com/guide/how-to-replace-the-prusa-nozzle-xl-multi-tool_506481](url) step 17
[image]

### How to reproduce

Install firmware, Go to Settings > Tools > Tool > Nozzle Diameter and set it to the corresponding size.
No nozzle diameter is shown.

CZDanol, 2024-08-23:
Confirmed. Internal ticket: BFW-5949

janecker, 2024-08-25:
Also just stumbled about this problem. Changed my nozzle and now I am unable to change it in the settings.

ClaGre, 2024-08-25:
> Also just stumbled about this problem. Changed my nozzle and now I am unable to change it in the settings.

Don't worry, be patient normally this problem will be fixed soon from dev team.
To work I've rollback the FW.

janecker, 2024-08-25:
> Don't worry, be patient normally this problem will be fixed soon from dev team. To work I've rollback the FW.

Yeah, hope so. Did not want to bother rolling back the firmware. Will ignore the warning for now (or might even disable it). Currently only one of the nozzles ist "wrong" so easy enough to ignore for now.

albjerre, 2024-08-26:
Same issue here om my Prusa XL dual-head.

ClaGre, 2024-08-31:
> > > Also just stumbled about this problem. Changed my nozzle and now I am unable to change it in the settings.
> 
> > 
> 
> > Don't worry, be patient normally this problem will be fixed soon from dev team. To work I've rollback the FW.
> 
> 
> 
> Which FW did you roll back? I am unable to install 6.0.4 due to "ESP error". Did you have the same error?

Absolutely not; I reinstalled 6.0.4 and everything works correctly (2 heads).
No ESP errors.

Have you tried reinstalling firmware from USB?
Have you tried to do an hard reset and after try to reinstall 6.0.4 from USB?

Karkovski, 2024-09-01:
> Absolutely not; I reinstalled 6.0.4 and everything works correctly (2 heads). No ESP errors.
>
> Have you tried reinstalling firmware from USB? Have you tried to do an hard reset and after try to [...]

I have tried a FW reinstall, but do not want to go through hard reset and full Selftest. I have 5T though.

Also, what is worth mentioning, I was seeing a lot of Dwarf-related errors after the update to 6.1.2 and these two issues (Dwarf and ESP errors) might be related.
However, I have made some progress in solving the issue, I think.
After disconnecting the XL Splitter (and effectively disabling T3-T5), the errors went away. When doing that (yesterday), I have taken out the XL Splitter board and cleaned the fuse blades (they were surprisingly, very much tarnished).
After that I did not get any Dwarf errors, but I have not tried downgrading to 6.0.4 yet, as I had to launch 30h print immediately after. Will try again when it's finished.

CZDanol, 2024-09-02:
Guys, this is a ticket about a nozzle diameter menu item. If you have issues with dwarves/ESP, please use a different ticket.

We should be publishing a fix release very soon. If you need to change the diameter before it is released, you can downgrade the FW, change, then upgrade back. Sorry for the inconvenience :(

danopernis, 2024-09-04:
Hi folks, v6.1.3 was just released and this should be fixed :rocket:

nazar-pc, 2024-09-06:
Confirm as fixed on dual-head XL

closed 2024-09-06 by danopernis