prusa3d/Prusa-Firmware-Buddy #4153: [BUG] Thermal runaway (fluke ?) on 6.1.2 on MK4S

vpelletier, 2024-08-24:
### Printer model

MK4S

### Firmware version

6.1.2

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

I got a thermal runaway error when printing. I believe this is a fluke, in the sense that it is probably not the primary issue. Nevertheless, given thermal issues are no jokes I prefer to report it.

Printer history: It is obviously very new. I received the kit version, and assembled it. The only notable issue I had was Y axis calibration failures, which were resolved by moving the Y axis bearings away from the printbed center. The assembly manual said to have them as close to the center as possible without overlapping with pockets ends - which is exactly how they were, but the calibration was not happy about it. Moving them maybe 1mm away from the center each was enough to pass Y calibration. Then I could print the provided benchy and fidget cube without any issue. Then I upgraded from stock 6.1.0 to 6.1.2 . The printer is on my LAN wifi.

After starting the 3rd print (the one which failed) from a freshly prusa-slicer-sliced [Maker's Muse clearance and tolerance gauge](https://www.makersmuse.com/view/downloads/clearance-and-tolerance-3d-printer-gauge), I started monitoring by accessing the printer's embedded website, using firefox 128.0.3 . I then switched tab to something else, and a few minutes passed with the printed progressing, besides me while I wasn't directly looking at it and with a headset on.

Until it stopped progressing: I first noticed what looked like the bootloader appearing on the screen, and soon after the red LED came up with a crash screen & the printer beeping: it detected a thermal runaway.

Here is what was on the print bed (reassembled after removal):

[image]

I hurried to the axis control menu and lifted the printhead, which extracted itself from the blob visible on the picture.

I peeled the failed print from the printbed.

I re-ran the full self-test suite, which passed without any issue.

On my browser, I switched to the tab on which I had the printer status, and it displayed the idle page. I switched away from that tab, and after a few minutes restarted the same print, from the same USB stick and the very same bgcode file. It is still going as I am writing these lines, having well exceeded the previous point of failure.

Here are the things which stick out to me:
- The apparent reboot: could it have been a more mundane (?) firmware crash which, upon boot, triggered a thermal runaway error (misdetection) ? This is the first time I see an error on a 3d printer, my Mini+ never had an issue in 3 years I've had it, so I have no idea what the crash process looks like (can it intermitently display a boot-like screen ?).
- Opening the embedded website and the changing the tab: Could the browser have suspended the tab ? Does firefox even do that, and to what extent ?
- The fact that it is the first print after upgrading to 6.1.2 failed, while previous prints succeeded. But I have no idea if this is even relevant: having only printed two small-ish jobs on 6.1.0 may not be enough to tell the printer is fine.

As I post this, the second printing attempt is still fine and has been running for 40 minutes.

bkerler, 2024-08-24:
We had similar issues with MK4 before, they all lead to a hw problem with the thermistor in the end. So if you have further issues, contact prusa support, they should be able to tell you if the thermistor is the root cause.

bkerler, 2024-08-24:
Also, the severe stringing might also indicate that your filament isn't dried, which might be another cause. Make sure it is dry.

vpelletier, 2024-08-24:
I just checked the hotend and thermar break thermistors (although only at room temperature) and both read 99k ohms, so they seem fine. They measure fine in the self test and when checking the stats page.

I wonder if the measurements taken by the printer are taken on the loveboard itself or if they get routed through the main loom. If it is the latter, it would be nice to have a pinout in order to check the mainboard end, in case there are issues in that loom. Which would be of course mostly pointless if the ADC is on the loveboard.

About the stringing, the filament should be still quite dry as I received it at the same time as the printer, and it is PLA (so AFAIK it should be slow to take in moisture). FWIW the benchy had minimal (though not zero) stringing.

One thing I discovered after completing the clearance test (2nd try went to the end without any new issue) is that my belt tensioning was way off. Not off enough to have the belt jump or calibration to fail, but still terrible. The reason I had issues using the app was apparently that the belts resonated at such low frequencies that it would not pick them up. I have no idea if bad belt tensioning could cause stringing, though. It most likely contributed to the clearance test to... Well, let's just say it "rolled a natural 1".

My impression on the incident, coming from the blob on the failed print, is that X/Y movement just somehow gave up. Maybe this could explain having the hotend buried in a large chunk of plastic, as I found it until I could lift it. Could this chunk cooling down have accelerated dissipation and tripped the thermal runaway ? As to how such glob can appear, I have no idea if it can happen on its own on a crashed printer (maybe just the heat melting some extra PLA when all motors are stopped while the printer rebooted ?) or if it needs the extruder to actually keep pushing (which could hint towards an X/Y issue as a further cause).

vpelletier, 2024-08-24:
Welp, it happened a second time, this time in the pre-heating of another print.

This time I jumped into the sensors pages, and bam: hotend measured at -23°C. Take cover off, wiggle thermistor wires, 170°C. Then -23, 169,etc, you get the picture: I have a broken wire on my thermistor.

And this time I was ready, and the boot screen happened again before the error - but the error did know this happened during preheat, so I'm now convinced that is the normal UI sequence when displaying an error screen.

closed 2024-08-24 by vpelletier

bkerler, 2024-08-25:
@vpelletier Yeah, I guess prusa has to improve quality inspection for the thermistors. Glad you were able to sort it out :)

Kammi61871, 2024-10-05:
Hi i have the same Problem on 6.1.3 bevor the upgrade to the MK4s there was no error, but the strange thing is the problem comes only when i printing PLA on PETG, ASA, ABS the is no Thermal error
[image]
[image]