prusa3d/Prusa-Firmware-Buddy #4155: Stealth Mode / lights should be choose-able in prusaslicer or prusa connect

ufgum, 2024-08-26:
### Printer model

XL 5 head

### Describe the enhancement

Describe the solution you'd like:

A switch to enable/disable "stealth mode" in prusaslicer before send the file to connect. Changing in connect would also be a solution.

Switch off all lights should also be an option, so it would be possible to make an "silent print" with one or two switches.

Many users has their printer in the living area. Switching always on the display is annoying.

### Expected functionality

Describe how it would work:
checkbox/es?

bkerler, 2024-08-30:
For requesting such feature in prusaslicer, please post an issue at the prusaslicer repo, which would likely require a specific gcode in the prusa firmware (not sure if such gcode already exists).

For prusa connect I assume such option would be easier to integrate and would make much more sense.

ufgum, 2024-09-02:
@bkerler 
Thanks, I did:
[PrusSlicer Repro](https://github.com/prusa3d/PrusaSlicer/issues/13324)

closed 2024-09-02 by ufgum