prusa3d/Prusa-Firmware-Buddy #4157: [BUG] M862 command not functioning over serial.

Java-SE-Binary, 2024-08-27:
### Printer model

MK4s

### Firmware version

6.1.2

### Upgrades and modifications

MMU3 Installed

### Printing from...

Octoprint

### Describe the bug

When sending the M862 command to the printer, specifically with the parameter 'P', the printer does not actually perform the appropriate check for the feature that the command passed. For example, sending 'M862.6 P "MMU5" does not trigger a check on the printer for a supposed MMU5 to exist, and trigger the appropriate warning prompt.

'M862.X P' is not documented to be disabled when printing over a serial connection, and I feel it would be beneficial to remain enabled during serial printing, similar to how the MK3 printers behaved.

M862.X Q seems to work as intended, with the exception of M862.6 Q. When passing M862.6 Q to my MK4s with Input Shaper enabled and a MMU3 unit installed, the printer reports "Input shaper" as the only installed feature, despite MMU3 being present.

### How to reproduce

Simply sending any variations of M862 with the 'P' parameter will show this behavior.
M862.1 P<nozzle_diameter>
M862.2 P<model_code>
M862.3 P"<model_name>"
M862.4 P<fw_version>
M862.5 P<gcode_version>
M862.6 P"<installed_features>"

### Files

[Cube_0.4n_0.15mm_PLA_26.4228_1h49m.zip](https://github.com/user-attachments/files/16755081/Cube_0.4n_0.15mm_PLA_26.4228_1h49m.zip)
This G-Code file has manually edited M862 checks that will cause an incompatibility prompt when printed directly on the printer. When printing through Octoprint, this prompt _does not_ appear.

Java-SE-Binary, 2024-11-11:
This is still an issue with firmware 6.1.3. Calling the M862.X P command from OctoPrint still causes no response, and M862.6 Q still replies with incomplete information.

danopernis, 2025-01-14:
Assigning myself to prevent stalebot from closing. This is not a priority, but I feel like it should be fixed at some point.

DerMouse, 2025-08-02:
Hello,
Any progress on this annoying point ? Just sent a 0.4-sliced-gcode file to a 0.25-equipped MK4 (via OctoPrint), and the printer happilly started printing without any warning.
Firmware is 6.2.4.

bkerler, 2025-08-25:
@Java-SE-Binary M862 is being used for FW features, not HW features. So MMU3 (or MMU5 or whatever) are HW features and these are reflected by the M862.2 model code.
This M862.6 Q just reporting "Input shaper" is expected behaviour, even though it should now report "Phase Stepping" for example as well. 

The M862.2 P implementation is somewhat weird and I will look into this as it should issue a warning if the printer isn't compatible....

Java-SE-Binary, 2025-08-25:
@bkerler I would expect M862.6 Q to return "Input Shaper" _and_ "MMU3" as looking to the default MK4S MMU3 profile within PrusaSlicer shows that it passes M862.6 P"MMU3" as well as M862.6 P"Input shaper" in the start gcode, indicating to me that it is trying to make the printer check that the MMU3 is enabled within firmware, especially for gcode that is sliced for an MMU3 printer. But, this is really a minor problem, the M862.X P commands are the bigger issue in my eyes.

All M862.X P commands that are passed to the printer elicit zero response on the printer's side (Outside of the response saying the command was received) when sent over serial. I'd expect _any_ of them to prompt an error on the printer, or back through the serial connection, if any of the values do not match what any of the M862.X P commands are checking for. All of these commands work just fine when running from a gcode file loaded onto a USB drive.

As far as I can see with the stock profiles within PrusaSlicer, none of the Buddy based printers use the M862.2 P command at this time, using M862.3 P instead.