prusa3d/Prusa-Firmware-Buddy #4159: [BUG] Entering temperature/fan/speed values through touch keypad does not work

Gyrxiur, 2024-08-27:
### Printer model

MK4

### Firmware version

6.1.2

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

When you enter temperatures, fan or print speed through touch keypad and confirm it, it changes in menu, but doesn't really change the value, if you go back to previous menu and return to menu with values, you can see they have not been changed. If you want them to really change, you have to double tap control knob, which then really changes to new value.

### How to reproduce

As in description

### Expected behavior

Values should just be REALLY changed after confirming them via touch keypad, not only changed on display.

### Files

https://youtu.be/N9P_kp6mAno

H0Z0, 2024-08-27:
This is already a known and confirmed bug. An internal ticket was opened. 
[#4149](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4149)

Gyrxiur, 2024-08-27:
WTF, why are you sending me Trojan:Win32/Leonem? BAN HIM NOW!

closed 2024-08-27 by danopernis