prusa3d/Prusa-Firmware-Buddy #4161: [BUG] Can´t change filament when one of the objects was cancelled

mattracing, 2024-08-27:
### Printer model

MK4

### Firmware version

6.0.4

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

If you are printing multiple objects with color change (M600) and you cancel one of them, when the execution of the filament change arrives, the firmware enters a loop and does not allow you to change it. You cannot continue with the impression.

### How to reproduce

Create an impression with 2 objects
Put a color change (M600) at the desired height (for example 0.6mm)
Start printing
Cancel one of the objects before color change

### Expected behavior

The change of filament will be executed
I will not unload the filament
You will have to remove it manually
You can't load the new filament whether you press Purge More or Retry

CZDanol, 2024-09-08:
Hey, could you please provide a sample gcode (ideally something small, like a 1cm cube), that reproduces this issue?

mattracing, 2024-09-24:
Until now I couldn't reproduce the failure with a cube. I usually print pet tags and it just had the same failure with the same version of Frimware and PrusaSlicer 2.8.1. I can't upload the GCODE of the failure because it has personal data. I was printing 4 pieces, two of them failed. I entered the object cancellation menu and cancelled them. When it finished completing the current layer, it ran the filament change procedure. However, the head just moved to one side and the screen popped up asking if the filament was unloaded. The motor didn't move at all. Since the filament was inside the nozzle, I couldn't remove it manually either. And if I tried to retry, the motor didn't move to unload it either. I only had the option of resetting and starting over.

closed 2024-12-02 by github-actions[bot]