prusa3d/Prusa-Firmware-Buddy #4162: [BUG] Lines that include # are treated as comments even when using quoted INI strings

Swizzler121, 2024-08-28:
### Printer model

MK3.5

### Firmware version

6.1.2

### Upgrades and modifications

None

### Printing from...

N/A

### Describe the bug

When filling out the prusa_printer_settings.ini if you are using a password (or any other key I assume) that begins or includes a # symbol, it's treated as a comment, even when in a quoted string (Feature added in latest update #3158).

Example config:
```
[wifi]
ssid = example ssid
psk = "#examplepsk"
```

I first noticed the issue in a previous firmware but had an alternative network at the time with a different psk that avoided the issue. When I saw the above mentioned feature I decided to try again, just to find this is still an issue.

This firmware update does include a way to manually enter the password using the screen, and using that you are then able to join networks that include a # in the login details, but compatibility with the prusa_printer_settings.ini file would be preferable.

### How to reproduce

Join a network with a PSK that includes a # sign by loading the prusa_printer_settings.ini file

### Expected behavior

It successfully loads the correct settings from the prusa_printer_settings.ini file and joins the wifi network successfully.

danopernis, 2024-08-28:
Yeah, I agree with you, this is a bug.

The thing is, the INI parser is based on https://github.com/benhoyt/inih and that is quite difficult to modify. Proper solution to solving this bug would most likely involve complete rewrite of the INI parser and that will take a lot of time, which would be better spent on solving other bugs and creating new features. Therefore, I don't think we will get to work on this anytime soon. If anybody wants to contribute proper parser based on proper tokenization and formal grammar, I would be happy to discuss details.

In the meantime, as a workaround, please use other means of entering the password, like the Prusa App or on-screen keyboard.

bkerler, 2024-08-28:
After debugging the issue, it seems the root cause is that inline comments are allowed. The issue is caused by this line: 
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/lib/inih/ini.c#L74

Imho, this doesn't make a lot of sense to allow inline comments in ini files and the easiest way to fix this is to just undefine INI_ALLOW_INLINE_COMMENTS [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/lib/inih/ini.h#L95). What do you think @danopernis ?

bkerler, 2024-08-28:
@danopernis This doesn't seem to be caused by the ini parser. The handler does interpret the value correctly as "#examplepsk" and does run [espif_join_ap(ap.ssid, passwd)](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/lib/WUI/wui.cpp#L327) with passwd being set to "#examplepsk".

bkerler, 2024-08-28:
@Swizzler121 it does seem that this is no bug. In fact, the quotation marks are treated as password letters. So if you define the password without quotation marks : 
```
[wifi]
ssid = example ssid
psk = #examplepsk
```
it should work as expected.

Swizzler121, 2024-08-30:
It did not work without quotes either, like I said i initially ran into the bug trying to use it as shown above without quotes, but when that failed I did some searching and saw the bug about allowing quoted lines, so I waited for that to be merged to try again.

bkerler, 2024-08-31:
@Swizzler121 I tried using your example ssid and psk and it works as it should with fw 6.1.2, it isn't treated as comment. Instead it treats quotation marks as password chars.

Swizzler121, 2024-09-01:
Is the a way to pull an error off the printer so I could see why it's failing in my case then? could be another unknown bug? I assumed it was the # as it was leading the password field and was the only special character, and occurred across two passwords that shared that trait even though they were being broadcast by different hardware.

bkerler, 2024-09-01:
Can you maybe send your .ini file via email ? It could be something else which does affect the ini parser. Or does your .ini file only contain this:
```
[wifi]
ssid = example ssid
psk = #examplepsk
```

bkerler, 2024-09-01:
Mhmm it seems that the space in the ssid is causing a possible bug.

bkerler, 2024-09-01:
Ok, I was now able to fully identify the issue and wrote a PR #4175 for it.
It seems that even if the same values are accepted with ini files with network field, in some variants the value fields are indeed misinterpreted as inline comments, leading to a nulltermination if the value starts with "#". This shouldn't be the case.

bkerler, 2024-09-01:
Sorry for causing so much trouble in the beginning, glad I finally was able to hunt down the issue properly :)