prusa3d/Prusa-Firmware-Buddy #4178: Wipe Tower: No Sparse Layers (EXPERIMENTAL) - can crash printhead when stuck filament is detected

kurtgluck, 2024-09-02:
### Description of the bug

Yes I know wipe-tower No Sparse Layers is  experimental.

XL went over to wipe tower, raised printbed, wiped, then decided this time it might have stuck filament - to be helpful it moves the print head to the front of the printer WITHOUT LOWERING IT!   Bash right into the model.   There is no way to recover from this.

Probably might be a good idea for the error routines:  "Stuck Filament", "Filament Out" etc to lower the print-bed.

dropbox link to video: https://www.dropbox.com/scl/fi/w4fa6281k38t0qs92c5pn/unifi_090124_1353_634.0s.mov?rlkey=4ibfy151b14o45655x1ynacju&dl=0

### Project file & How to reproduce

[sinkv3.3mf.zip](https://github.com/user-attachments/files/16830183/sinkv3.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-x64

### Operating system

MAC 14.6.1 (23G93)

### Printer model

Prusa XL 5TH

murk-sy, 2024-09-02:
Duplicate (but stale), contains some extra information:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1461

kurtgluck, 2024-09-03:
From a firmware perspective.  All ( :-) ) that needs to be done is for the XL to keep track of the LOWEST (call this “L” ) the printbed has been during the current print and whenever any of the error states that present the toolhead to the front of the bed occurs for the XL to first lower to 1) record where it is currently (call this “C”) 2) move the the aforementioned LOWEST (L) position 3) after error condition is handled by the user and the print head moves back to resume … move back to the height before this all happened “C”.

closed 2025-11-20 by github-actions[bot]