prusa3d/Prusa-Firmware-Buddy #4181: Branch 6.1.2 missing

lezetuser, 2024-09-03:
### Question

I miss in in the FW the build tree 6.1.2 branch. Does it mean that the master is in synch with 6.1.2 or is it generated outside the public folders? When  last time I built the master (the 6.1 was a official release) the fw did not work on my MINI.

danopernis, 2024-09-03:
Hi @lezetuser you are looking for the tag, not a branch.

Current policy is to always tag the releases with `vMAJOR.MINOR.PATCH` and to keep `master` branch pointing to the latest MINOR release + community pull requests which were merged after the release.

All of these inconveniences stem from the fact that a lot of the development is happening behind closed doors, which has some marketing reasons which I don't really want to talk about :smile: 

We are planning to sum this in the documentation and also describe the release and contribution processes, it just takes time :disappointed: 

Out of curiosity, what didn't work for MINI?

lezetuser, 2024-09-03:
it stucked in the middle of flashing

lezetuser, 2024-09-03:
when I look, the last public tag is also only 6.1.0

closed 2024-09-03 by lezetuser