prusa3d/Prusa-Firmware-Buddy #4182: [FEATURE REQUEST] Use empty tool if available for bed level calibration

dzwiedziu-nkg, 2024-09-04:
### Printer model

XL/Multi tool

### Describe the feature

Bed level calibration cause filaments dots on plate. I have XL with 5 tools. In most cases one or more tool are empty. 

### Expected functionality

Use empty tool if available for bed level calibration before printing instead tool used for printing.

ClaGre, 2024-09-05:
This is a good analysis, I also found this thing and even if it did not prejudice the prints it seems right to me to avoid contaminating the printing plan with these points of contact.
Good idea 👍🏻

dzwiedziu-nkg, 2024-09-05:
I just discovered that I can use in start g-code in Prusa Slicer:

`; select tool that will be used to home & MBL`
`T4 S1 L0 D0`

instead 

`T{initial_tool} S1 L0 D0`

Then the fifth tool will be used.