prusa3d/Prusa-Firmware-Buddy #4184: [BUG] MK4 does not show object preview after update to firmware v. 6.1.2 and 6.1.3

ThomiBon, 2024-09-06:
### Printer model

MK4

### Firmware version

6.1.3+7898

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

After updating my MK4 from firmware 6.0.4 to 6.1.2+7894 and now to 6.1.3+7898, printer stopped showing object images while printing.

Settings in Prusa Slicer under "Printer" -> "G-code-thumbnails" are: 16x16/QOI, 313x173/QOI, 440x240/QOI, 480x240/QOI, 640x480/PNG

(I did not change these values neither before nor after the FW update.)

### How to reproduce

Send gCode via Prusa Connect, start printing. While printing, you can chose between 2 views (either screen with buttons to stop, pause etc. or screen showing print details. No object image is shown any more.

### Expected behavior

Printer screen starts showing a object thumbnail image after several minutes of printing and returns to showing image as default screen state after touching display or pressing kbob to see print details (as was before firmware update).

danopernis, 2024-09-06:
Hi @ThomiBon just to be sure, can you post a screenshot of "Settings -> User Interface -> Print Progress Screen"?

ThomiBon, 2024-09-06:
Settings -> User Interface -> Print Progress Screen
[image]

danopernis, 2024-09-06:
So just to be sure, even after 30 seconds, the image is not shown?

ThomiBon, 2024-09-06:
Exactly - no image during the whole printing process.

ThomiBon, 2024-09-06:
@danopernis, I exported the gCode, transferred it to the USB stick directly (i.e. I put stick in my laptop) and now an image is shown! In all other cases when image was not shown I used Prusa Link or Prusa Connect for file transfer.

danopernis, 2024-09-06:
That's actually quite helpful! I wasn't able to reproduce it, but now I might be. Thanks :)

larrrox, 2024-09-07:
I was able to re-produce the issue too, this is happening to me over Prusa connect and just regular USB transfer.

H0Z0, 2024-09-08:
Try to set the print progress screen higher than 30 seconds. (For example 31 seconds.)  Based on my tests, it looks like 30 seconds doesn't work, but with the >30 seconds setting it's fine.

ThomiBon, 2024-09-09:
@danopernis, @larrrox : Same finding as described by @H0Z0 above (thanks a lot for mentioning and confirming, I found out about it as well!). All "print progess screen" times > 30s work, not depending on whether file is sent by Prusa Link, Prusa Connect or direct USB. The other way round, a setting to 30s does leads to no image displayed.

crasbe, 2024-09-09:
I can confirm the issue is present on my Prusa XL (currently still at version 6.1.2) and after setting the timeout to 31s the image is shown.

I think I found the bug. The minimal value for the timeout is set to 30s in this line:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/src/gui/MItem_tools.hpp#L582-L592

However when the stored timeout is compared to the minimum value (probably to abdecken the "none" case), it checks if the configured value is BIGGER than the minimum, which it is not when set to 30s:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/src/gui/dialogs/print_progress.cpp#L84

For reference: "getTime" returns the stored print progress timeout value:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/src/gui/dialogs/print_progress.cpp#L129-L133

A fix could be to change L84 in "print_progress.cpp" to a ">=" instead of ">". But I currently have no means to test that fix here.
```diff
PrintProgress::PrintProgress(window_t *parent)
-   : DialogTimed(getTime() > MI_PRINT_PROGRESS_TIME::config.min_value ? parent : nullptr, GuiDefaults::RectScreen, 1000 * getTime())
+   : DialogTimed(getTime() >= MI_PRINT_PROGRESS_TIME::config.min_value ? parent : nullptr, GuiDefaults::RectScreen, 1000 * getTime())
    , estime_label(this, Rect16(text_left_side_offset, text_baseline_y, left_column_width, text_label_height), is_multiline::no)
    , estime_value(this, Rect16(text_left_side_offset, text_baseline_y + text_label_height + text_value_y_offset, left_column_width
#if HAS_LARGE_DISPLAY() // adding middle_column_width because middle column is currently unused & the left estime_value doesn't have enough space to hold the current version of all the data
                                 + middle_column_width
#endif
                             ,
                             text_value_height),
          is_multiline::no)
    , info_text(this, Rect16(text_left_side_offset, text_baseline_y + text_label_height, info_text_width, text_value_height), is_multiline::no)
    , progress_bar(this, Rect16(Left(), GuiDefaults::ProgressThumbnailRect.Height(), Width(), GuiDefaults::ProgressBarHeight))
    , progress_num(this, Rect16(Width() - progress_num_x_offset, progress_num_y_baseline, progress_num_width, progress_num_height))
    , thumbnail(this, GuiDefaults::ProgressThumbnailRect, GuiDefaults::OldSlicerProgressImgWidth)
    , more_icon(this, Rect16(Width() - icon_x_offset, icon_y_baseline, icon_size_x, icon_size_y), more_icon_res)
    , gcode_info(GCodeInfo::getInstance())
    , time_end_format(PT_t::init)
    , mode(ProgressMode_t::PRINTING_INIT) {
```

larrrox, 2024-09-17:
Changing the print progress timeout to 31 seconds or 32 seconds did not seem to fix the issue on my MK4 (6.1.3)

crasbe, 2024-09-19:
> Changing the print progress timeout to 31 seconds or 32 seconds did not seem to fix the issue on my MK4 (6.1.3)

On bigger models, it takes several minutes for the preview to show for the first time on my machines (like 5-10 minutes). After that it will consistently show after the set period.
Does the preview not show for you at all, even on longer prints?

acothebraco, 2024-09-20:
I have the same problem on both of my Printers. One is MK3.5 and the other one MK4S

CZDanol, 2024-09-20:
Hello. Thank you for reporting the issue, we will work on fixing it :)

Internal ticket: BFW-6067

larrrox, 2024-09-27:
Thumbnails have seemingly started working again once I re enabled them today with newly sliced files. I have print progress screen set to 32 seconds. Previously it wasn't working with the identical settings.

acothebraco, 2024-09-27:
> Thumbnails have seemingly started working again once I re enabled them today with newly sliced files. I have print progress screen set to 32 seconds. Previously it wasn't working with the identical settings.

Thank you. I changed the screen timeout also to 32 seconds and the preview works.

yasars, 2024-10-07:
in the alpha 2 not working,  nothing helped!! use Connect / Link...and Mk4s

CZDanol, 2024-10-07:
I've checked the code and there some weird things, like conditions that the nozzle needs to be at least 1 cm above the printbed for the progress screen to show and such. I guess additionally to the 30s bug found by @crasbe , this can be contributing to it as well. Should show up eventually though.

closed 2025-01-09 by CZDanol

crasbe, 2025-01-09:
@CZDanol thank you for fixing this and thank you for mentioning me in the Release Notes :)

CZDanol, 2025-01-10:
@crasbe thank you for making the job easy for us :)