prusa3d/Prusa-Firmware-Buddy #4185: [BUG] Recalibrating X- and Y-axis before every first print after power up

muesli, 2024-09-06:
### Printer model

MK3.5

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

Since the 6.1.2 firmware update, the printer seems to recalibrate the X- and Y-axis before every first print after being powered on. Is this intentional?

Other users are reporting this happening on the Mk4 as well: https://www.reddit.com/r/prusa3d/comments/1f4npp9/mk4_calibrates_x_and_y_axis_before_every_print/

### How to reproduce

1. Turn the printer on (so it does a fresh cold start).
2. Print something.

### Expected behavior

Avoid unnecessarily re-calibrating either axis.

me25542, 2024-09-07:
I have also experienced this, idk if its intentional.
In addition to calibration before each print, the printer also recalibrates the axis on each "Auto home".
As I see it , printing worked OK with just homing and no calibration before each print, but maybe there is some good reason for this. 🤷

muesli, 2024-09-07:
The prints are fine indeed. I guess I'm really just curious if this is intended behavior or an issue manifesting itself.

rangido, 2024-09-12:
I am also experiencing exactly the same thing, but with my MK4.

JetroDR, 2024-09-15:
I have experienced the same issue since 6.1.2. Previously it would just home both axis by hitting the sides twice. But now it goes on for about a dozen times.

shofeur, 2024-09-16:
I am also experiencing this on my MK4S (upgrade) on the first print after power on but I only noticed it after I updated to 6.1.3.  It is tolerable if the behavior is intentional, but I didn't notice anything in the release notes regarding a change such as this.

CZDanol, 2024-09-17:
Hello, I've created an internal ticket BFW-6069 for this.

mix579, 2024-10-02:
It certainly feels excessive how often the axes are getting calibrated.

Stefan-Odw, 2024-10-18:
Same problem with my MK4.

Bazimek, 2024-12-10:
Same issue for my MK4S.

Belott, 2024-12-10:
> Hello, I've created an internal ticket BFW-6069 for this.

It seems that there are several users with the same problem. My MK4 has this behaviour to. So, what happened with the internal ticket for this problem?

danopernis, 2024-12-13:
Hi, I checked the internal ticket. It looks like this was confirmed on our side and it is just waiting in the queue with other tickets. We do not consider this to be a critical bug, although I can understand that it can be annoying. We will eventually fix this. Assigning myself to keep stalebot from closing.

Stefan-Odw, 2025-01-11:
Firmware 6.20 RC1 ist out and still nothing concerning this nagging issue?
Really?

muesli, 2025-01-11:
I can confirm this issue still persists with 6.2.0-rc1.

Belott, 2025-01-12:
I don't understand why Prusa doesn't solve this problem. Many users keep reporting this issue and I can't imagine it's that difficult to solve.

rbm78bln, 2025-01-17:
...and causes unnecessary wear out...

Belott, 2025-02-15:
Hi CZDanol, do you have any news about this issue?

fsvelim1, 2025-02-15:
Please take care of this, we are all waiting. It is anoying. I think it also damages the printer slowly. Even upgrade to MK4S did not help.

CZDanol, 2025-02-15:
The fix is coming up in 6.2.1

ErwanAliasr1, 2025-02-18:
@CZDanol would be cool to get a link to this issue for reading the associated commits and learn more about the internals

mix579, 2025-02-18:
Received my flat Core One yesterday, and as hard as it is to imagine, the banging at the beginning of a print—and in particular the first one after booting—feels even worse than on the Mk4S. Maybe it's just my imagination because the enclosure is such an awful resonator.

Belott, 2025-02-19:
I installed 6.2.1 straight away yesterday. I've done a homing around 10 times since then. Result: error fixed, works! I hope it's permanent... It's a bit unfortunate that it took so long for Prusa to take the error seriously. But at least it seems to have been fixed now. Thank you @CZDanol for persistently pursuing this error! You were the last glimmer of hope...

CZDanol, 2025-02-19:
Okay, glad it works for you guys. Closing the issue.

@mix579 Core ONE being a CoreXY printer has a completely different homing procedure and its banging is unrelated. If you want to discuss that, let's open a different issue.

closed 2025-02-19 by CZDanol

LoakAudio, 2025-03-06:
Hi all, sorry to intrude on the discussion.
I've not tested 6.2.1 but 6.2.2 on an MK3.5 still do the full calibration, hitting the zero position 16 or 18 time after a cold start.
Issue does seems completed for me.

CZDanol, 2025-03-06:
Hello @LoakAudio - this happens after every cold start?

LoakAudio, 2025-03-07:
Hi @CZDanol I've tried several times a print from a cold start and it did the full calibration again BUT, the idea came to me to do  an "XY calibration" from the "Calibrations & tests" menu of the printer and since then, no more full calibration on cold start.
A reset of the "system" calibration was probably needed.
Sorry I didn't think of that earlier.

CZDanol, 2025-03-07:
I see. Glad it got resolved then :)

mix579, 2025-04-08:
> Okay, glad it works for you guys. Closing the issue.
> 
> [@mix579](https://github.com/mix579) Core ONE being a CoreXY printer has a completely different homing procedure and its banging is unrelated. If you want to discuss that, let's open a different issue.

In case anyone sees this and wants to add their voice, it's issue #4582