prusa3d/Prusa-Firmware-Buddy #4186: [BUG] Enclosure option keeps switching off

zuidwijk, 2024-09-07:
### Printer model

XL 5 tool head

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

USB

### Describe the bug

When a print start (via connect) I get the error message that enclosure fan isn't working.
When I check te settings, the enclosure toggle is off in the settings.
When enabling it starts spinning and prints fine.

The next print same issue, enclosure is switched off again

### How to reproduce

Just start a print, it happens every time ... I didn't switched off the printer, its always on.

### Expected behavior

When enabling the enclosure in the settings I expect it to remain on, until I switch it off.

zuidwijk, 2024-09-08:
Did three prints today, each time it was off and I had to re-enable the enclosure.

Is it a Slicer thing? That the slicer turns it off?

zuidwijk, 2024-09-08:
https://github.com/user-attachments/assets/e12e48c8-b850-4d17-a42a-5110850f9013

This is what's happening

zuidwijk, 2024-09-08:
https://github.com/user-attachments/assets/c5de301d-4597-4b26-9c89-2e5ed1200adb

3rd time today...

zuidwijk, 2024-09-17:
> This is not a bug, this is a correct behaviour. After the printer detects that the fan is not spinning, it turns enclosure off and displays this message. Did you check the fan for debris and inspect the wiring as the dialog suggests?

When I enable the enclosure you hear the fan spinning up and is working properly. 

So that raises the question in my mind, why isn't the fan starting when the print starts, but does start/work when I re-enable the enclosure?

zuidwijk, 2024-09-17:
It happened again. I started the print, you year spinning up the fan for a second and than turns of with the error message. 

Turning it on on the display and you hear spinning up the fan and it's just working.

reify3d, 2024-09-18:
I can confirm that this happens to my XL5 as well. Running 6.1.3 FW. No debris in the fan or filter. Fan cover removed in case it was too tight like the print fan on the toolheads causing issues. Did not resolve the issue.

danopernis, 2024-09-18:
Well, for some reason, printer thinks the fan is not spinning. You still haven't ruled out electrical/mechanical problem. There might be some problem on the tachometer wire. I suggest consulting tech support, unless you are willing to grab oscilloscope and build your own firmware to debug this problem. Just to be clear, I believe you that your enclosure is misbehaving, I just don't think this is a firmware problem given that there are not many reports like this from other users.

zuidwijk, 2024-09-21:
> @zuidwijk did a factory reset and updated to the 6.2.0-alpha FW and the issue is resolved. Hope it works for you too.

No not yet, I'll try within a few days. I'm just not into alpha firmware on production environments... I'll sit this one out till next release. At least my suspicion looks in the right place.

zuidwijk, 2024-09-21:
> Well, for some reason, printer thinks the fan is not spinning. You still haven't ruled out electrical/mechanical problem. There might be some problem on the tachometer wire. I suggest consulting [...]

Yeah I hear you. Haven't had time yet. Way too busy. But I'll contact support anyway for extra check-up.

zuidwijk, 2024-09-25:
Well I’m in contact with support via email. Here another one. Checked if enclosure was enabled, which it was, started a print and again the notification.

ngolf, 2024-10-01:
I am having the same issue. Enclose fan works perfectly, passes the check when I turn the enclosure on, but when sending a print through prusa connect, I get that message. If I start a print from the printer, no issue. Re-enabling the enclosure immediately after getting that error message turns the fan on.

zuidwijk, 2024-10-04:
Running 6.2.0 alpha1 and later alpha2 fixes the issue for me. So it is firmware related, I'm not going back to 6.1.3. 

The 6.2.0 alpha2 also handles "old/wet" filaments better. I've got obxidian hardened nozzles and now I can select that in the settings. 
[image]

ngolf, 2024-11-02:
I think the issue happens when running a filament that doesn't need filtration (e.g. PETG). I don't seem to get the issue when running PC-CF, for instance.

Tholi78, 2024-11-17:
Same Problem here, started recently.
Wiring is OK and nothing is blocking the fan. So i'll try the alpha releases.
Thanks @zuidwijk for investigating

Mrjohns42, 2025-01-18:
> I think the issue happens when running a filament that doesn't need filtration (e.g. PETG). I don't seem to get the issue when running PC-CF, for instance.
> 
> This is as if the printer checked if the fan was running, regardless if there had been a command to set the fan on or off, so if fan/filtration not requested for that filament, the printer sees that the fan is not running, and complains.

I think you're onto something.  I seem to only get this when running PETG.  Enclosure fan and wiring seem fine.

freeformz, 2025-01-20:
I get this when running PLA. Fan runs fine when I re-enable the enclosure from the "Tune" menu.

closed 2025-03-31 by github-actions[bot]