prusa3d/Prusa-Firmware-Buddy #4188: [BUG] Missing layer pieces when printing from prusa connect sometimes

larrrox, 2024-09-08:
### Printer model

MK4

### Firmware version

6.1.3

### Printing from...

Prusa Connect

### Describe the bug

Sometimes when printing via Prusa Connect I get missing layer sections. I can print the same thing via USB directly and the issue does not happen. Even if I reprint via prusa connect the issue never happens in the same place/time. The same thing is happening in these forum posts [https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-general-discussion-announcements-and-releases/prusa-mk4-missing-lines-randomly/paged/2/](url) [https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/missing-layer-sections/](url)

### How to reproduce

It's seemingly random behaviour, but it always happens when printing from Prusa Connect.

### Expected behavior

Printer should not skip sections of a layer ever

### Files

Sorry for the lack of pictures, I threw out most of the prints, I'll attach more when it happens again. In this image, it appears it skipped part of one of the previous layers and then printed over it. 
[image]

murk-sy, 2024-09-09:
Duplicate, see also link for some basic steps to try and mitigate it (hard to say if it'll help though):
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3682#issuecomment-1998040694

Also just to clarify, has the issue actually occurred on 6.1.2 or 6.1.3? 6.1.2 Added binary gcode checking so it so knowing that should help the devs

larrrox, 2024-09-12:
> Duplicate, see also link for some basic steps to try and mitigate it (hard to say if it'll help though):
> 
>     * [[BUG] MK4 skipped printing a part mid layer  #3682 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3682#issuecomment-1998040694)
> 
> 
> Also just to clarify, has the issue actually occurred on 6.1.2 or 6.1.3? 6.1.2 Added binary gcode checking so it so knowing that should help the devs

Occurred on 6.1.3

larrrox, 2024-09-12:
Seems like the only way to fix it is by using G-Code. Only seems to be happening on BGcode streaming through prusa connect, might be wrong though. Noticed that it never happens when I re-print the same object after it's already happened.

closed 2024-11-19 by github-actions[bot]