prusa3d/Prusa-Firmware-Buddy #4189: [[BUG] Prusa MK3.5 display does not show filament change info anymore

AWSW-de, 2024-09-08:
### Printer model

MK3.5

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

The Prusa MK3.5 display does not show filament change info anymore in the automatically scrolling display infos. Now it just shows the n/a text although there are several color / filament changes in the print that was sliced with Prusa Slicer 2.8.0.

On previous versions < 6.1.2 the info was shown. 

Thanks a lot in advance for bringing this useful info back. =)

### How to reproduce

- Slice a model and add color changes in Prusa Slicer 2.8.0.
- Send the sliced model to Prusa Connect.
- Start the print on the Prusa MK3.5.
- Check the display information for remaining time, ETA and the filament change info. 

### Expected behavior

The filament change info should be shown.

gudnimg, 2024-09-08:
There is a known PrusaSlicer bug in 2.8.0.  If the printer doesn’t receive the correct g-codes then the printer won’t show this info. 

See: https://github.com/prusa3d/PrusaSlicer/issues/12849

One of the users in that thread mention MK3.5 being affected too.

AWSW-de, 2024-09-08:
Thanks a lot! Not found this topic before posting. 

I will close this here then therefore. 

Thank you =)

closed 2024-09-08 by AWSW-de