prusa3d/Prusa-Firmware-Buddy #4190: [ENHANCEMENT] Repeat Print should remember Filament Mappings

BrienKing, 2024-09-09:
### Printer model

XL

### Describe the enhancement

Right now if your select "Reprint" and the last print re-mapped the filaments since the order of the filaments changed you have to remap the filaments again.

### Expected functionality

I want reprint to do exactly what it just did.  If I remap filaments it should save that information and just reprint.

danopernis, 2024-09-17:
I can easily see how implementing this would annoy some other user who would want to actually do the toolmapping again on reprint. But perhaps we should at least default to previous toolmapping so the workflow is more smooth.

BrienKing, 2024-09-17:
Yes, I don't mind if it goes to the Tool Mapping screen again, as long as my previous mappings are there.

CZDanol, 2024-09-18:
Internal ticket: BFW-6084