prusa3d/Prusa-Firmware-Buddy #4191: Choose what you want to Preheat when using the Preheat function.

bluerider098, 2024-09-09:
### Printer model

XL

### Describe the feature

Currently preheat turns everything on, including all print heads.  When you then print a file any print heads not used will remain heated throughout the entire print. 

The workaround is to use the control tab and turn everything on manually, but it might be a nice feature to select what you want to preheat. 

### Expected functionality

When you select preheat you have a screen come up similar to the filament screen when your start a print, and select which print heads you'd like to preheat.

danopernis, 2024-09-16:
This is more involved. I recently found out that Prusa Slicer even emits the code to disable heating of unused tools, however after the toolmapping screen is dismissed the unused tools are disabled and all the G-codes related to them are ignored. We will likely solve this particular use-case by cooling down the unused tools in some future version of the firmware.

Other than that, what are the use-cases for selecting specific tools for preheat?

CZDanol, 2024-09-16:
Seems like a duplicate of #1795 . We already have an internal ticket BFW-5879 and I'd like to hop to it when I have enough free time.

closed 2024-09-16 by danopernis

bluerider098, 2024-09-16:
@danopernis While this is a similar request, I'm not seeing specifically what I'm asking for here listed on 1795.  I'm specifically asking to not preheat all tools, just the ones you intend to use.   So can this be added to that?

CZDanol, 2024-09-16:
@bluerider098 I've noted it down as a request, we will definitely consider it

bluerider098, 2024-09-16:
The use case is there is no reason to preheat tools you don't intend to use.  Wear and tear on a printhead that your not using.