prusa3d/Prusa-Firmware-Buddy #4193: [BUG] /api/v1/job GET response contains invalid JSON

ysmilda, 2024-09-09:
### Printer model

Mini+

### Firmware version

>= v5.0.0-alpha3

### Printing from...

PrusaLink

### Describe the bug

When a job is present without `stat` the returned JSON response for a GET request to `/api/v1/job` is invalid. It contains a comma at the end of the object which should only be there if the message has `stat`.

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/lib/WUI/link_content/basic_gets.cpp#L434-L438

Moving the `JSON_COMMA` macro from the end of line 434 to the start of line 436 should fix the message.

danopernis, 2024-09-16:
Thanks for the contribution, this is scheduled for next 6.2.0 version

danopernis, 2024-10-01:
@ysmilda I pushed this to the 6.2.0-alpha2 which was just released, in case you want to try it before final 6.2.0 is released.

CZDanol, 2025-02-20:
Release is out, closing :)

closed 2025-02-20 by CZDanol