prusa3d/Prusa-Firmware-Buddy #4194: First layer issues when using Complete Individual Objects on subsequent parts.

bluerider098, 2024-09-09:
### Printer model

XL

### Firmware version

6.1.2

### Upgrades and modifications

None

### Printing from...

USB

### Describe the bug

Using the Complete Individual Objects I often get first layer issues on some of the parts printed after the first one.  I believe the issue is related to the fact that the shape of the bed changes over time as it heat soaks, and possibly from the adjacent completed objects possibly changing the shape of the bed, or possibly the bed mesh is not being applied to subsequent parts. 

### How to reproduce

Run any print where the 2nd part is 40 or so minutes after the first.  The nozzle will be too far off the bed for the first layer. 

### Expected behavior

Not sure what the issue might be as I didn't know how the bed mesh is applied to these type of prints.  

Possibly a new bed mesh is created in the area of the subsequent parts before it's started.  I realize that oozing and the fact that the mesh area may need to be smaller to avoid hitting the already printed parts makes this solution far from ideal.

Another better option might be to apply a nozzle offset that's applied to any parts printed after the first. 

I don't have a good answer, but it's only an issue when using Complete Individual Objects.  If the parts are printed all at once the first layers are good. 

I'm my particular case the parts that I am printing have a very tight tolerance and the parts come out different sizes when printing as an individual vs. printing a group of them.  I have narrowed the issue down to the fact that the amount of cooling time between layers is much greater when printing multiple parts vs. a single part.  So by using the Complete Individual Objects feature my problem is solved because the cooling between layers is always the same.

danopernis, 2024-09-17:
@bluerider098 can you please provide minimal G-code which exhibits this behaviour and photos of the failed first layer?

bluerider098, 2024-09-17:
I unfortunately don't have any pictures, but will rerun a print tonight and get some pictures as well as share the gcode.

bluerider098, 2024-09-18:
@danopernis Attached are some pictures exhibiting the issue. You can see how the front sides of the parts did not adhere well to the bed. One of the objects 

The other picture shows a 1st layer that I stopped that was not good.  Here also is a link to a video as it was printing. 

https://youtu.be/eTh87xrT6aE?si=GmFg8CvzWfxgB-fC

Now if I run this exact same group of parts, but turn off complete individual objects it'll print just fine.

[image]
[image]
[image]
[image]
`
[image]
[image]
[image]
`

bluerider098, 2024-09-18:
[image]
[image]
[image]
[image]
[image]
[image]

bluerider098, 2024-09-18:
> Hmm, there are different temperature settings for the nozzle and the bed for first and non-first layers. Do these differ for your profile? Could be that the first layer settings is not applied for subsequent objects, or that the nozzle/bed do not heat up/cool down fast enough for the further first layers.

Yes I'm not really sure.  The only thing I change in the slicer settings is the complete individual objects.  That was my reason for starting this issue because I'm scratching my head on trying to figure out how to make this work.  I've had some success by lowering the live Z while the first object is printing, but that's been hit or miss on completing good parts. 

I think it's that the shape of the bed changes as parts are printed and everything warms up, but that's just a theory. I don't know enough about how the printer uses the bed mesh etc to analyze it any further.

bluerider098, 2024-09-19:
@CZDanol I think you're onto something with the different bed and nozzle temperatures for the first layer. 

 I did more testing and made the heatbed temperature the same for the first layer as the rest of the print and did the same for the nozzle temperature.  Doing this I've done 2 successful test prints.  Tonight I'm going to run another test print, but much longer, approx 14 hour long print to see how that performs.

bluerider098, 2024-09-21:
I've run into another potential improvement for sequential printing when you have a tool changer. 

So the issue is that when you have a Toolchange that happens at any time before the already printed parts are below the tool head clearance.  Essentially the printhead will clear all of the other parts for printing, but not if a tool change drags it back over the already printed parts.  In my case the tool change was at the start of the next part and it ended up trapping the printhead between an already primed part and the dock.  Because crash detection is not working yetthe print head was bouncing off the already printed parts and the dock. It sounded horrible.

This issue could be easily fixed by lowering the bed by the inputted extruder clearance before a tool change, and holding that position until the tool change is complete.  Then raising back up and then continuing to print. 

 I'm going to try and figure out how to write custom g code to make this happen, but in my opinion this should be standard operating procedure for sequential prints that have Toolchanges.  In my case the tool head clears all the other parts, but not if a tool change happens at anytime before all of the other parts are below the print head.

CZDanol, 2024-09-22:
Thanks, I've passed the idea over to the slicer team, we will have a discussion about it hopefully :)

bluerider098, 2024-09-30:
@CZDanol 

I'm kinda stuck for finishing this particular test.  Do you have any G-Code writing knowledge? 

I need some help creating G-Code to move the print bed down 10mm ( relative move, so 10mm down from where it's at) , do a Toolchange, then raise the 10 mm and resume printing.  I've tried inserting G-Code in the tool change section in PS, but I didn't really know what I'm doing and I've had 1 crash and another where the z value seemed wrong after the Toolchange and printing resumed. 

I think my issue is that the printer i believe is working in absolute mode, and I need it to move relative just for the Toolchange.  I could be wrong but that's my guess.

bluerider098, 2024-10-04:
@CZDanol-prusa 

So after more testing I can confirm that if you leave the first layer bed temperature the same as the printing temperature the problem is resolved.  I also recommend leaving the nozzle temperature the same as well, as the nozzle will also change at different temperatures. 

I'm convinced that the shape of the bed is different depending on temperature which was leading to the failures. The bed mesh was being created at the first layer temperature and then as the temperature changed so did the shape of the bed essentially making the bed mesh incorrect.

So id recommend a warning in PS when complete individual objects is selected if your first layer temperature differs from other layers. 

Should I create another issue for lowering the bed at Toolchanges when using complete individual objects, or is what I put here good enough?

CZDanol, 2024-10-04:
I'd rather guess that the bed temp simply lowers after the first object and that causes first layer adhesion issues

bluerider098, 2024-10-04:
> I'd rather guess that the bed temp simply lowers after the first object and that causes first layer adhesion issues

I'm my case printing PETG the bed temperature actually was higher for other layers, 90 vs. 85

closed 2025-10-27 by github-actions[bot]