prusa3d/Prusa-Firmware-Buddy #4197: [BUG] Print stopped and failed after 1 day, stubs.cpp:5. "USB or file error" before BSOD

jeremiahbiggs, 2024-09-10:
### Printer model

XL Multi Tool (5 toolheads) with Enclosure

### Firmware version

6.1.3+7898 XL [S] [A]

### Upgrades and modifications

Prusa Enclosure

### Printing from...

USB Drive

### Describe the bug

**Problem:**
Print stopped after one day of printing. Job continued on the screen, but did not progress and did not throw error. Could not pause and resume job. 
Troubleshooting steps:
- Initiated filament change to try to get job going again.
- Attempted pause/resume of job, job was stuck on "resuming" for about 10 minutes. 
- Simulated power outage (unplugged printer during job)
- Printer picked up job and returned to previous "stuck" state. As before, screen indicated print job was still going, but print heads did not move and did not make progress.
**New Error, same print issue (no printing):**
"USB Drive or file error, the print is now paused. Reconnect the drive."
Troubleshooting steps:
- Pulled USB drive, checked gcode with Prusa Slicer gcode viewer, application did not map/display print job correctly. Replaced gcode file on USB drive with new one from computer that mapped correctly in Prusa Slicer gcode viewer.
- "USB Drive or file error..." message reappeared
- Simulated power outage again
- Job resumed, same issue with no progress and then "USB Drive or file error" message displayed.
**New Problem:**
Received BSOD stubs.cpp:5
[image]
[image]
[image]

### How to reproduce

I have not and will not try to reproduce this issue, it is too costly in filament. Will try other prints, I have had other successful prints before this.

### Files

[2024-09-10.zip](https://github.com/user-attachments/files/16948382/2024-09-10.zip)
https://1drv.ms/u/s!AvvWcWNFM39FicYv_polQH7djmslQw?e=eK1fGp

jeremiahbiggs, 2024-09-10:
Link to Gcode: https://1drv.ms/u/s!AvvWcWNFM39FicYyZawn0Ve4Pv7Wug?e=O9a7jb

danopernis, 2024-09-16:
@jeremiahbiggs thanks for the report and especially for the crash dump.

The error stems from incorrect usage of `StreamRestoreInfo` in [`set_restore_info()`](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/v6.1.3/src/common/gcode/gcode_reader_binary.hpp#L37). This particular bug [is fixed in v6.2.0-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/v6.2.0-alpha1/src/common/gcode/gcode_reader_binary.hpp#L38-L43) as part of the general fixes to media prefetch subsystem.

We are probably not going to release v6.1.4 so please try using v6.2.0-alpha1 or wait for the v6.2.0

closed 2025-12-16 by github-actions[bot]