prusa3d/Prusa-Firmware-Buddy #4198: [BUG]Android Prusa App Reports Wrong Firmware Version

phos2lampsei, 2024-09-11:
### Printer model

ML4

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

Prusa Connect 

### Describe the bug

After upgrading the firmware on my Prusa MK4 from 6.1.2 to 6.1.3 the Android Prusa App is misreporting the firmware as the older 6.1.2 version instead of the current 6.1.3 version. 

Prusa Connect from within Prusa Slicer 2.8 does report it correctly although where it appears in Prusa Connect within Prusa Slicer did seem to change between firmware versions 6.1.2 and 6.1.3. The screen on the printer itself also reports the 6.1.3 firmware version correctly. This appears to be some sort of bug with how the firnware layout was changed between versions that in particular is impacting the Android Prusa app. 

### How to reproduce

Open Prusa app on Android.
Click on the idle Prusa MK4 printer
The displayed firmware version is incorrect

### Expected behavior

I expect the displayed firmware version to match the actual firmware version.

danopernis, 2024-09-17:
Hi, thanks for the report. This repository is for bugs related to firmware running on the printer, not the Prusa App. I forwarded your report to them, since it appears they don't have any public issue tracker at the moment. I am closing this issue now and hope they will resolve it soon on their side.

closed 2024-09-17 by danopernis

phos2lampsei, 2024-09-17:
Thank you for the reply!

To further clarify, the issue with the Prusa app ONLY occured after the
printer firmware upgrade to 6.1.3. The app properly reported the firnware
version immediately before the upgrade to printer firmware 6.1.3 when the
printer was still on firmware 6.1.2.

The printer firmware upgrade to 6.1.3 was the triggering issue for the app
issue.

These facts surrounding the issue suggest that either a bug in the firmware
caused this issue or that a planned firmware change with 6.1.3 was not
properly accounted for and coordinated with the app.

Thanks,
Scott