prusa3d/Prusa-Firmware-Buddy #4199: [BUG] WiFi 'link down' regression

flynny75, 2024-09-11:
### Printer model

XL 5TH

### Firmware version

6.2.0 alpha 1

### Printing from...

N/A

### Describe the bug

WiFi fails to connect, network status just shows 'link down'. I have tried entering the credentials manually with no effect. Searching for networks in the WiFi wizard finds no networks. Printer is physically fine, was working on 6.1.3

RPospisil, 2024-09-11:
Having the same issue on XL. Wifi connection is unstable - on wifi status page I see 100% ping loss mostly. 
Rarely wifi looks like connected, pings looks ok, but printer is offline in prusa connect.
I downgraded to 6.1.3 and wifi works.

liyanage, 2024-09-13:
I'm seeing the same, installed 6.2.0 alpha 1 on my XL 5T today and WiFi no longer works. I see two failure modes:

* The network status either says "Link Down" as mentioned above, or "No AP"
* When I am in that first state, I can power cycle the machine and it will connect, including acquiring the IP address via DHCP, and it will properly resolve buddy-a.connect.prusa.com to an IP address. However, even in this state, it will not be reachable on the network. Ping from another host on the same network just goes nowhere.

As a workaround, I had to connect an Ethernet cable.

fogarasia, 2024-09-13:
I have the same problem, WIFI drop down. The previous version worked well.

VibroAxe, 2024-09-14:
I'm also experiencing this. Possibly related to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4200

danopernis, 2024-09-16:
Thanks for reporting, we are aware of the issue and we are tracking this internally as BFW-5957

VibroAxe, 2024-09-16:
> Thanks for reporting, we are aware of the issue and we are tracking this internally as BFW-5957

Great, any diagnostics we can usefully collect?

danopernis, 2024-09-16:
Not really, but thanks. Just to provide some context:

There have been several changes to Wi-Fi subsystem and all of them were supposed to improve the performance, but it turned out they actually worsen it in some cases. As a hotfix, we are going to revert all of these changes and keep Wi-Fi as it was in previous versions.

Changes involved updating ESP SDK to newest version, rewrite of the parser for the protocol used to communicate with ESP and CRC check for said protocol. We will try applying these changes separately to better understand which of them causes this regression.

kchristensen, 2024-09-25:
Also experiencing weirdness with regards to WiFi -- Is there a way to "unset" WiFi credentials w/o a settings reset? I've wired my printers and would prefer to just "forget" the WiFi settings but I haven't figured out how to do so.

mikelane, 2024-09-29:
Experienced this today as well.

danopernis, 2024-10-01:
Hi folks, v6.2.0-alpha2 just got released which contains the revert of the Wi-Fi changes. You might want to re-test this.

ethanrae, 2024-11-05:
Wi-Fi is working for me again in v6.2.0-alpha2

CZDanol, 2024-11-05:
Closing as done.

closed 2024-11-05 by CZDanol