prusa3d/Prusa-Firmware-Buddy #4200: [BUG] Prusa Connect Status Error - Refused

kayblasto55, 2024-09-12:
### Printer model

Prusa XL 5 Tool

### Firmware version

6.20-alpha1

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

after update Prusa connect says OFFLINE, on the XL under setting for Prusa Connect Status "Error" and under error "Refused".

### How to reproduce

Update to 6.20-alpha1 and refuses to connect to Prusa Connect

murk-sy, 2024-09-13:
This may be unrelated, but symptoms are similar to what I had at some point

Are you using wireless or wired connection? Did you ever use one and then just the other (for example: first used wireless and then ran a cable and now using wired)?

kayblasto55, 2024-09-13:
Using wireless, never have used wired. Prusa Connect was working, did the update and it stopped working, downgraded to 6.1.3 works. I updated again and stopped working so I downgraded again and works again all I have done is the update haven’t messed with any setting.

stigsknudsen, 2024-09-13:
I have the exact same issue.

VibroAxe, 2024-09-14:
I also have this issue, tried toggling wifi/eth/wifi to get it to reconnect and now 6.2 wont connect either

EightBitAlex, 2024-09-15:
Having the same issue on this firmware. Tried a lot of things. I eventually downgraded to 6.1.2. and that connected and worked right away.

816-80085, 2024-09-16:
Had same issue since upgrading to 6.1.2 on 5h xl using wifi. cannot use connect - I reverted to 6.1.3 and connect works again

danopernis, 2024-09-16:
This might be a problem with Prusa Connect and this might be a problem with Wi-Fi. It is hard to tell based on the info in this issue. It would help if you could try reproducing this on Ethernet. It would also help if your reports would include the printer model.

r3Fuze, 2024-09-16:
Everything works as expected when using Ethernet.

When using Wi-Fi the network status is either `No AP` or `Link down`, so I believe it's a Wi-Fi issue, and not related to Prusa Connect.

Printer model: Prusa XL (5T)

fliedlr, 2024-09-29:
using WIFI only upgraded from 6.1.3 to 6.2.alpha1 same issue refused connection to Prusa XL and prusa connect where as my MK4S and MK3.5 have stable connection also using the same Wifi.
Printer XL 5T
Mk4S ( 6.2 alpha1) , MK3.5( 6.1.3), Mini

closed 2024-12-09 by github-actions[bot]