prusa3d/Prusa-Firmware-Buddy #4203: [ENHANCEMENT] MMU3 spool join and filament run out enhancement

majedation, 2024-09-12:
### Printer model

MK4 with MMU3

### Describe the enhancement

Requested enhancement:
When the MMU detects that the filament has run out, the printer continues printing till the extruder filament sensor detects that the filament has run out (it keeps printing with the filament between the MMU and the extruder).
After the extruder detects filament run out, the MMU loads the filament from another slot (in case of spool join), or the printer asks to load new filament (in case only one filament is loaded to the MMU), then it resumes print.
This way, the 30-40 cm filament between the MMU and the extruder is used, and also prevents the filament from getting stuck during unload, this happened with me; the end bend in the filament got stuck in the MMU during unload, and the printer stopped printing and showed an error message.

### Expected functionality

MMU detects filament run out -> printer continues printing -> extruder detects filament run out -> printer stops and new filament is loaded (without unloading the previous filament since it's been completely used) -> printer resumes printing.

Mike-FUT, 2024-10-02:
I also didn't see the point so far why the remaining filament was pulled out from the nextruder instead of just slowly pushing the new one through. Especially with the MMU modifications of the Nextruder the filament run our is detected really late when the filament already left the upper idler gear of the nextruder. Very little filament left in the Nextruder at this point in time.
So when this happens why not extrude the remaining 5 millimeter to a dump until the Nextruder gears no longer transport anything and then load the new filament from MMU to Nextruder gears and slowly push that through the nozzle?

Mike-FUT, 2024-10-02:
> Prusa doesn't poop like bambu

It has to, otherwise filament change when running out wouldn't be possible because you can't start printing with a freshly loaded filament without purging first (unless you want unfilled areas in your print).
And right now it does it anyway after the spooljoin filament change.
This change requested by @majedation would mean only an extra bit of filament on the dump.

fyby, 2024-11-11:
This functionality would be great! I dont have any problem with throwed out filament but with filament stuck while unloading. Tried two times and two times I had to intervene to be able to finish print.

Regarding "pooping". I think when extruder detects end of filament, it can purge rest into wipe tower together with loading next spool. That should not be a problem. I dont know, but unloading is really a problem.

CZDanol, 2024-11-11:
I don't think we can implement this easily. If there would be a color change between the runout and the spool join activation, MMU would not be able to pull the filament out. We would have to also consider how much filament is needed till the next color change/print end/whatever.

Regarding pooping - we indeed do need less purging after filament change thanks to our fine-tuned tip forming procedure.

majedation, 2024-11-11:
@fyby 
The print can be only in 1 color without a wipe tower.
I suggest doing it this way:
After the new filament is loaded, it purges the extra filament into the print itself between the infill gaps, or something close to this idea, it will be hidden inside the printed part, it will not be visible.
Or after purging in the air, the nozzle slightly touches the edge if the part while returning to continue printing, only a tip of the extruded filament will be attached to the printed part, and that could be cleaned by a sharp tip.
Anyways it already purges in the air now, I think there will no issue in only making the filament not unload but load the new filament to the previous one without unloading.

majedation, 2024-11-11:
@CZDanol 
In that situation (color change between runout), the the extruder can extrude the remaining filament in the extruded (but not detected by the MMU) it can extrude it into the wipe tower, then load the new filament.

Regarding purging after spool join, since the extruder already extrudes filament out in the air after spool join (single color print, wipe tower not activated), then it doesn’t matter how much it extrudes out, the purged filament will be already in the air, it will not hurt extruding a little bit more.

CZDanol, 2024-11-11:
We cannot extrude into wipe tower - wipe tower is generated statically in the slicer, the printer cannot change it dynamically. We would need to purge into thin air, and we would then need to also purge into thin air after the filament change, because the purging volume would need to be much higher.

fyby, 2024-11-12:
You are right. It would be possible only when you configure spool join for that print in slicer. Only then the slicer would be able to project wipe tower with reserve for potential spool join. But that would make wipe tower much bigger imho. So it doesnt make sense.

kabelkao, 2024-11-28:
It would be nice to get notification that filament run out and spool join is active. I can clean purged filament from print. Sometimes falls into prints. Its pain to wipe in air cause purged filament is joined to printing object and 1st new layer is thin (maybe due not balanced pressure on nozzle?).