prusa3d/Prusa-Firmware-Buddy #4204: [BUG] XL Multitool Drops toolhead with new right side home in 6.2 RC

Reesesben2, 2024-09-14:
### Printer model

XL Multitool

### Firmware version

6.2.0 RC1

### Upgrades and modifications

None

### Printing from...

USB

### Describe the bug

When XL Multitool Homes it now homes tool selector to right side however it can do this while a tool is atached which drops the current tool since the Selector Arms extend past where the tool selector bumps to home

### How to reproduce

home the Prusa XL with a tool in the selector

### Expected behavior

printer should home on left side only and not go all the way to the right when a tool is in the selector

Reesesben2, 2024-09-14:
I'm unsure if this is relevant but it happens after doing a cold pull

danopernis, 2024-09-17:
Thanks for reporting. I was able to reproduce this using following steps from control menu
1. pick tool 1
2. move x axis all the way right and move y axis to 200mm
3. auto home
4. repeat steps 2. and 3. until the tool is dropped (it doesn't always happen)

I have created internal ticket for investigation.

CloudStrife1983, 2024-12-31:
I just had exactly the same problem. For me it was as follows:
1. printing stopped during probing.
2. print head moved to the rear right and moved too far to the right and so the print head unlocked
[image]

o-vesely, 2025-03-04:
Hi,

I just now tried to reproduce it on the current firmware version (6.2.1), but wasnn't able to. If you're still experiencing this issue on the latest release, please let us know with any additional details so we can investigate further. 

Does it happen consistently or just sometimes?
How many toolheads do you have on your XL?

Looking forward to any feedback!