prusa3d/Prusa-Firmware-Buddy #4208: [BUG]

andrewgynous, 2024-09-15:
### Printer model

XL

### Firmware version

6.2.0-alpha1+8672

### Printing from...

prusa connect

### Describe the bug

When adjusting first layer z height during print printer freezes and resets.

### How to reproduce

start print, then hold down selection knob to bring up nozzle height adjustment screen, change height.  at this point the freeze/ reset happens

danopernis, 2024-09-17:
I tried this and wasn't able to reproduce it.

andrewgynous, 2024-09-17:
Happened to me three times in a row with two different print files.  I'll try refreshing the firmware and see if that eliminates the issue.

andrewgynous, 2024-09-18:
> I tried this and wasn't able to reproduce it.

After same firmware reload issue does not occur

closed 2024-09-18 by andrewgynous