prusa3d/Prusa-Firmware-Buddy #4211: [FEATURE REQUEST] Add PVA to the list of supported filament materials under Load Filament

jp-phys-NEIU, 2024-09-16:
### Printer model

XL Multi, MK4 MMU

### Describe the feature

Add PVA to the list of supported filament materials while doing a Load Filament, and set PVA load/unload temperature to 205C. Alternatively, provide "Generic" or "Other" designation with a user-settable load/unload temperature. Currently, user must specify a material type during Load Filament. PVB is a close substitute, but when selected, the firmware reports a mismatch between the PVA entry in the .gcode file and the PVB firmware selection. 

### Expected functionality

List of materials would include PVA, and would set the nozzle temperature to 205C for loading and unloading operations.
This would eliminate the warning message when firmware detects the difference between PVA specified in the .gcode file and non-PVA material selected during Load Filament (since PVA is not an option currently).

CZDanol, 2024-09-17:
Hello, user filaments are already done in 6.1.3: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.1.3

CZDanol, 2024-11-26:
Closing as done - doable through user filament.

closed 2024-11-26 by CZDanol