prusa3d/Prusa-Firmware-Buddy #4215: [FEATURE REQUEST] PrusaLink HTTPS Support

Redstonecrafter0, 2024-09-19:
### Printer model

MK4S

### Describe the feature

HTTPS would make PrusaLinks authentication secure as the credentials would be encrypted.
Secure namespaces like .dev would work to access to 3D printer. I use a .dev domain locally. At least Google Chrome does not allow http access on secure namespaces.

In an IoT vlan without internet access Prusa Connect would be no alternative.

### Expected functionality

The certchain and private key could be imported from the USB flash drive. Alternatively they cloud stay stored on the flash drive and be loaded on boot since when HTTPS support is desired one would upload via HTTPS anyway.
If there is no user provided (trusted) certificate there should be a untrusted one generated.

Redstonecrafter0, 2025-01-26:
Hi, I appreciate your time and effort regarding this issue.

However I have trouble understanding the following:
- Prusa Connect is already using TLS 1.2 between the printer and the cloud. [Prusa Blog](https://blog.prusa3d.com/prusa-connect-a-network-solution-for-remote-control-of-3d-printers_90364/)
- TLS 1.2 for an embedded HTTPS server would be challenging because of power and memory constraints.

In my eyes this is the same problem to solve where one of them is already solved. 

If I am missing something, I'd really appreciate your insights. Thanks again!

Redstonecrafter0, 2025-01-29:
I took the liberty of testing the server using wrk. With more than two concurrent connections the requests/sec reduces. It's true that browsers open several connections at the same time for faster loading but in a local network it does not matter since TCP SYNs are retried and the latency is negligible.

The option for security should not be sacrificed for speed. It should at least be tested in a live environment if an HTTPS server would be viable. Otherwise I only see a self-hosted instance of Prusa Connect as an option which I highly doubt will happen.

Redstonecrafter0, 2025-08-21:
Can I at least convince you to implement this because of the Cyber Resilience Act?

moritzmhmk, 2026-08-29:
Prusa Connect Local https://www.prusa3d.com/p/prusa-connect-local might make this issue obsolete - at least for my use case 🤔

JHam215, 2026-10-12:
I was perusing feature requests and noticed this one.  HTTPS isn't already supported?  @moritzmhmk, that link you shared for prusa-connect-local is now 404.  However, a new link for early access is up at https://www.prusa3d.com/p/prusa-connect-local/early-access/
I'm a little confused why local connect is a service you have to request access to, but anyway....
Also, not stale.