prusa3d/Prusa-Firmware-Buddy #4216: [BUG] PrusaLink shows wrong temperature after print finishes

michaelrommel, 2024-09-20:
### Printer model

XL Dual Extruder

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

PrusaSlicer via PrusaLink

### Describe the bug

minor inconvenience, just if s.b. already works on that part of the code, it could be corrected. Immediately after the print finishes, the displayed temp abruptly goes to 6degrees... It observed it twice after a print with ABS.

[image]

### How to reproduce

- Slice with PrusaSlicer
- upload and start print via PrusaSlicer
- open PrusaLink in a browser
- let the print finish and observe the reported temperature

### Expected behavior

It should continue to report the actual temperature

michaelrommel, 2024-11-20:
If nobody cares or it is already fixed and people are to overwhelmed to close issues, so be it.

KlausKraemer, 2025-01-13:
[image]

As you see, the nozzle temperature in Prusa Link (XL 5 Tool) is nonsense. As I am going to start a print, the bed temperature is about the room temperature. So a nozzle temperature of 6°C is crap.

And on a XL 5 Tool, there are 5 nozzles. Which one is shown and where are the other's temperatures? (and please don't refer to the diagram!)

Currently using FW 6.1.3+7898

zehkaiser, 2025-01-15:
I would also like to see all 5 tool temps on the UI, even if they're hidden behind a mouseover.

closed 2025-03-24 by github-actions[bot]