prusa3d/Prusa-Firmware-Buddy #4218: [BUG] Y-axis fails with FW 6.2.0-alpha1, but passes with FW 6.1.2

Printer606, 2024-09-21:
### Printer model

MINI+

### Firmware version

6.2.0-alpha1

### Upgrades and modifications

Filament Runout Sensor; Improved Fan Duct

### Printing from...

USB Drive with Prusa Connect setup

### Describe the bug

Y-axis fails with FW 6.2.0-alpha1, but passes with FW 6.1.2

### How to reproduce

Flash FW 6.2.0-alpha1, then run XY Axis in Selftest.

encip, 2024-10-02:
I have this same issue with 6.2.0-alpha2 on my MINI+

The Y-axis calibration fails after moving fully to one extent then over to the other extent.

The calibration passes fine with earlier stable FW versions, and printing works fine as well.

encip, 2024-12-02:
Issue persists with 6.2.0-alpha2, latest available for Mini+.

Fretkaguru, 2024-12-16:
Same here. works on stable fails on 6.2.0-alpha1

encip, 2025-01-09:
Issue persists with 6.2.0-RC1

japato, 2025-01-11:
Test XY passed. It was HW issue. Manual moment range of Y was -3 to 179 on my MINI. After 1mm backward drift front left bearing range increase to 180 and now is all Ok. Seems It was the same problem solved on MK4 Y bearings space.
But primary installation Y axis bearings in case of Mini was done by Prusa team.

encip, 2025-01-11:
Just did some testing aligned with your comment.  I disabled the motors and manually pushed the carriage to full extents and can get full range -2 to 180 with manual motor control, but Y calibration still fails at around 179.5.

Visual inspection of the bearings shows them appearing to be properly centered.

What changed with the Y calibration in 6.2?  Why does this same setup pass with 6.1.2 and earlier but fails starting with 6.2.0-alpha1?

danopernis, 2025-01-15:
Hi folks, thanks for the report. This is most certainly not intended. I have created internal ticket BFW-6617 for investigation and try to push this for 6.2.0 final.

CZDanol, 2025-01-30:
Hello, 6.2 has been released: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0

The issue should be addressed, so closing. Have a nice day! :)

closed 2025-01-30 by CZDanol