prusa3d/Prusa-Firmware-Buddy #4220: Text with Pause command not shown in printer display

KimA998, 2024-09-23:
### Description of the bug

After you have sliced your project and have inserted a pause in the print - the text you have typed is NOT shown in the printers display on my MK3.9. 

Another issue/bug is when the printer has paused you have three options: tune, resume and stop. The stop bottom is selected by default!!! IT should be the RESUME.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Win11

### Printer model

MK3.9 with MMU3

nkappler, 2024-09-26:
I have a MK4S and noticed the same problem.
The text is actually shown on the display, but only for about 1 second or so.

RodKimmell, 2024-10-15:
It's overwritten by the Time Remaining messages.  Time remaining should be suspended until user responds to the Pause options.

RodKimmell, 2024-12-18:
This exact problem existed in the Einsy RAMBo board for the MK3.  It was fixed in that firmware.
Kindly assign this task to help get it resolved.

CZDanol, 2024-12-19:
Kindly closing this as a duplicate of #3143

closed 2024-12-19 by CZDanol