prusa3d/Prusa-Firmware-Buddy #4222: [BUG] Bed Leveling RETRY option can damage print bed surface.

RAMilewski, 2024-09-24:
### Printer model

Mini

### Firmware version

6.2.0-alpha1, 6.1.2, 6.0.4

### Upgrades and modifications

None

### Printing from...

Octoprint 1.10.2

### Describe the bug

My mini has begun failing bed leveling about 10% of the time.  (Still diagnosing the problem.  The knowledge base article only mentions other models).

After a failed leveling attempt there are two on-screen options.  RETRY and CANCEL.  If I choose the RETRY option the printer DOES NOT retry bed leveling, but instead begins the print, but with zero (or slightly less) z-offset which leaves the nozzle scraping along the bed.  

With firmware 6.2.0-alpha1, the message notifying the user of the failure also contains a QR code.  This contain the url ofr a knowledge base article on how to diagnose and fix the problem, but instead is the url for a page touting the Mk4s. (?!?)

### Expected behavior

The printer should make another attempt to level the bed.

If using 6.2.0-alpha1 firmware the QR code should contain the url of a Knowledge Base article providing help diagnosing and fixing the failed bed leveling.

RAMilewski, 2024-09-25:
Apologies:  I misquoted the screen.  Here's a screenshot of the bed level failure notice from v6.1.2:

[image]

CZDanol, 2024-09-26:
We've failed to reproduce the issue on our Minis. I'll try to direct our support guys to you to potentially do some diagnostics.

RAMilewski, 2024-09-26:
Does that mean that after a failure to level the bed your minis will actually retry the bed level operation?    How are you inducing a failed bed level on a properly working printer?

michalrudolf, 2024-09-27:
We created a build, that induced mesh bed leveling error on purpose and tried it on MINIs with 6.2.0-alpha firmware. Printer always attempted to level the bed again.

Prusa-Support, 2024-09-28:
Sounds like the PINDA may be partially damaged and falsely triggering in the wrong place subsequent times. I'm just thinking out loud though.
We didn't get similar reports and we can't reproduce the problem so, so far, it may be the case of a failed PINIDA probe but I must kindly ask you to contact our Customer Support for a thorough troubleshooting session before taking further action.

The troubleshooting will serve in either case to collect as much information as possible about the problem - which may be useful for the purpose of reproducing the error - or confirm a hardware failure.
https://help.prusa3d.com/article/customer-support_2287 

Feel free to write an email to my attention with reference to this issue (my name is in the signature) but I'm kind of positive that any agent may collect sufficient information to identify hardware faults even in a simple live chat troubleshooting session.
I seem to understand this issue is not even reproducible all the time but please try to have as much evidence at hand to start the troubleshooting, especially videos of the problem if you can catch it.

Michele Moramarco
Prusa Research

RAMilewski, 2024-09-30:
> We created a build, that induced mesh bed leveling error on purpose and tried it on MINIs with 6.2.0-alpha firmware. Printer always attempted to level the bed again.

I'm curious as to how you induced the bed level error.

RAMilewski, 2024-10-02:
Was able to reproduce the problem. Video below.
[image]

https://github.com/user-attachments/assets/3f0a9f74-6f4a-4c14-b0eb-ced28b451125

Note that the printer does NOT retry the bed leveling operation.

If you listen carefully, you can hear the nozzle scraping the bed at 2:29 into the video.

I'll contact Michele to resolve the problem in the printer, but the firmware is clearly not working properly.

CZDanol, 2024-10-03:
Oh, I haven't noticed that you're using Octoprint before.

RAMilewski, 2024-10-09:
Closed pending more testing without Octoprint.

closed 2024-10-09 by RAMilewski