prusa3d/Prusa-Firmware-Buddy #4225: [BUG] Wifi Link Down , no wifi possible,    MK3.9S   FW6.13

beng27, 2024-09-29:
### Printer model

MK3.9S

### Firmware version

6.13

### Upgrades and modifications

None

### Printing from...

PrusaLink / Connect

### Describe the bug

Upgraded from 6.12 to 6.13 , now printer just displays 'link down' for wifi. Entering manually does not fix. Bizarrely, my router can see the printer, has found its host name correctly, and allocated it an IP address. But the printer is not visible to Slicer or Connect by Wifi. Must be a problem in the WiFi stack onboard. 

### How to reproduce

Try to connect by wifi

### Expected behavior

able to send connect by wifi

beng27, 2024-09-29:
Annoyingly, after downgrading to FW 6.12 , the problem persists. Does the downgarde not replace the WIFI module software?

beng27, 2024-09-29:
So still on FW 6.12,  I put the old wifi module in (the one from the 3.9 prior to upgrade to 3.9S) and wifi works fine.
Conclusion: 
EITHER my new wifi module is suddenly faulty, concidentally at the same time as the 6.13 FW upgrade
OR the FW6.13 wifi code is buggy and the downgrade flash to FW 6.12 does not replace that.

danopernis, 2024-10-01:
Looking at the [changelog](https://github.com/prusa3d/Prusa-Firmware-Buddy/compare/v6.1.2...v6.1.3) there doesn't seem to be any change related to Wi-Fi or networking between these two particular versions.

beng27, 2024-10-01:
OK,  I will re-flash and see if the issue disappears

beng27, 2024-10-01:
Reflashing with 6.13 and the issue has gone away, so I am closing this

closed 2024-10-01 by beng27