prusa3d/Prusa-Firmware-Buddy #4226: [BUG] cancel object causes color change fail

spblat, 2024-09-29:
### Printer model

Mk4s, Mk3.5s

### Firmware version

6.1.2

### Printing from...

USB or Connect

### Describe the bug

When printing more than one instance of a print that includes a filament change (not MMU), canceling one or more instances before the color change makes the print fail

Use case: making several copies of an object that involves manual color change(s). But at print time I might want to print fewer objects than what is in the gcode. 

### How to reproduce

- Import any model in PrusaSlicer
- Apply a height-based color change 
- Add one or more additional instances of the model to print
- Arrange, slice and send to printer

Before the color change happens:
- Use control panel to “cancel object”
- Cancel one or more of the instances, leaving one or more left to continue printing
- Wait for color change
- BUG: printer will fail to eject filament when it’s time for a color change. Printer will prompt to manually replace filament by opening the idler but ultimately cannot resume print

### Expected behavior

After canceling one or more instances, the color change should proceed normally

cfillion, 2024-12-13:
I've had this happen today (MK4 6.1.3+7898). After an object has been cancelled, subsequent color changes skips the "Unloading" step and jumps to "Was filament unload successful?" instead. I had to forcefully remove the filament.

The insertion step is also broken. It skips straight to "Is color correct?". The "Purge more" button does nothing: it loops back to "Is color correct?" immediately without the extruder motor moving at all.

I had to reset the printer to get the firmware unstuck. Then it promptly resumed the print but without complaining that no filament was loaded... Thankfully it did that from the previously cancelled object (which surprinsingly the printer did not remember was cancelled after reset!) so I could salvage the rest of the print job via "Change filament".

...or so I thought: the X axis shifted a bit during the above causing the rest of the print to be misaligned by ~0.5mm... :cry:

CZDanol, 2024-12-16:
Internal ticket: BFW-6473