prusa3d/Prusa-Firmware-Buddy #4227: [QUESTION] Error when compile prusa mini firmware

Tiinpii, 2024-09-29:
### Question

Hello, i got this error when compile in python 3.9
I already update certificate but still got that error . some one know how can i fix this 
PS D:\Prusa-Firmware-Buddy> python utils/build.py --preset mini,xl --generate-bbf
Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.
PS D:\Prusa-Firmware-Buddy> Downloading gcc-arm-none-eabi-13.2.1
Traceback (most recent call last):
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\http\client.py", line 1257, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\http\client.py", line 1303, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\http\client.py", line 1252, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\http\client.py", line 1012, in _send_output
    self.send(msg)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\http\client.py", line 952, in send
    self.connect()
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\http\client.py", line 1426, in connect
    self.sock = self._context.wrap_socket(self.sock,
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\ssl.py", line 1040, in _create
    self.do_handshake()
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Prusa-Firmware-Buddy\utils\build.py", line 837, in <module>
    main()
  File "D:\Prusa-Firmware-Buddy\utils\build.py", line 798, in main
    bootstrap.bootstrap()
  File "D:\Prusa-Firmware-Buddy\utils\bootstrap.py", line 290, in bootstrap
    install_dependency(dependency)
  File "D:\Prusa-Firmware-Buddy\utils\bootstrap.py", line 202, in install_dependency
    download_and_unzip(url=url, directory=installation_directory)
  File "D:\Prusa-Firmware-Buddy\utils\bootstrap.py", line 145, in download_and_unzip
    f, _ = urlretrieve(url, filename=None)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 239, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 517, in open
    response = self._open(req, data)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 494, in _call_chain
    result = func(*args)
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "C:\Users\tiinp\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 1349, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>

danopernis, 2024-10-01:
And what is the question :sweat_smile: ? The build scripts tries to download dependencies and your python doesn't like the SSL certificate. You should try figuring which dependency this is and fix your python or system certificate store or something completely different.

larrrox, 2024-10-03:
try re-install python

closed 2024-12-10 by github-actions[bot]

AndreasArendt, 2025-01-14:
this is linked to #4407 and I have fixed this and created a pull request for this in #4408