prusa3d/Prusa-Firmware-Buddy #4228: [BUG] 6.2.0-alpha1 Firmware Prevents Wifi from connecting at all

mikelane, 2024-09-29:
### Printer model

XL 5 Tool

### Firmware version

6.2.0-alpha1

### Upgrades and modifications

None

### Printing from...

PrusaLink or Prusa Connect

### Describe the bug

## Description
Printed successfully from PrusaConnect on my XL 5-tool last night. Decided to flash to 6.2.0-alpha1 this morning before I began another print. Flash was successful. Printer was immedately shown to be offline on PrusaConnect. Tried several times to connect to PrusaConnect via the printer interface. Result error was `Refused`. Tried to connect via PrusaLink, printer could not be found. Double checked the wifi settings on the printer interface, showed as connected at IP 192.168.111.243. Used my modem's app to search for the prusa-xl client, no such client or any client at 192.168.111.243. Used netstat to check the status of 191.168.111.243, said "Host seems down". 

Tried connecting with ethernet by plugging in the cable and selecting ethernet in the printer interface. Got connected with IP 192.168.111.118. Was able to connect to PrusaConnect successfully. Was given the option of flashing back to 6.1.3, and chose to do so. After 6.1.3 was flashed, I selected wifi as the network to use in the printer interface. PrusaConnect continued to work. Used PrusaConnect to start a print successfully.

## Slicer Info
- PrusaSlicer version 2.8.1+MacOS-arm64

## Start Config
- Firmware 6.1.3
- Wifi connected
  - prusa-xl device available via modem admin page at 192.168.111.243
- PrusaLink available
- PrusaConnect connected and can print

## Problem Config
- Firmware 6.2.0-alpha1 (flashed via PrusaConnect)
- Wifi displays as connected
  - Connection says IP is 192.111.168.243
  - Unable to find prusa-xl device on modem admin page
  - `netstat -sP 192.168.111.243` resulted in "Host seems down."
- Unable to find printer using PrusaLink
- Connection Refused for PrusaConnect
- Connected printer using Ethernet
  - Set ethernet as network device on the printer
  - Connected successfully, IP 192.168.111.118
  - No device found on PrusaLink at 192.168.111.118 (or anywhere else)
  - PrusaConnect was successful
- Was given the option to flash to 6.1.3 from the PrusaConnect console 

## Ending Config
- Firmware 6.1.3
- Switched network back over to wifi
  - Connection says IP is 192.111.168.243
  - prusa-xl device available via modem admin page at 192.168.111.243
  - `netstat -sP 192.168.111.243` results in "Host is up"
- PrusaLink connected successfully
- PrusaConnect connected and print started from PrusaConnect interface in PrusaSlicer

### How to reproduce

- Use an XL 5-tool printer
- Verify you can connect and print via wifi, PrusaLink, and PrusaConnect on firmware 6.1.3. 
- Use PrusaConnect to flash to the 6.2.0-alpha1 firmware
- Determine if printer is available on the network, PrusaLink, or PrusaConnect

### Expected behavior

Expected behavior is for the Wifi networking to be operational after the 6.2.0-alpha1 firmware has been flashed.

mdjurfeldt, 2024-09-29:
I had the same problem yesterday.

mdjurfeldt, 2024-09-29:
Probably the same as issue #4199

CZDanol, 2024-09-30:
Closing as a duplicate.