prusa3d/Prusa-Firmware-Buddy #4229: [BUG] Enclosure fan not spinning when starting print

ngolf, 2024-09-30:
### Printer model

XL 5T, with original prusa XL enclosure.

### Firmware version

6.1.3+7898

### Printing from...

Prusa Connect

### Describe the bug

When starting a print from prusa connect, I get the following message about the fan not spinning.

[image]

The "enclosure" setting on "tune" is then automatically turned off.

If I turn back on, I see a brief message about checking the fan, and the fan starts normally.

Connections are OK, there is nothing blocking the fan, and I can feel the air flow at fan exit, as well as see the fan RPM on telemetry.

### How to reproduce

Send a print to the printer (enclosure setting must be on)

JCDwight, 2024-10-01:
I have the same error.  I have to walk to the printer every time and turn it on if I want it on.  This just started happening recently.

danopernis, 2024-10-01:
@ngolf @JCDwight thanks for reporting, this seems like a duplicate of #4186

Please add your comments to #4186 and close this one in order to keep the issues more organized.

ngolf, 2024-10-01:
duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4186

closed 2024-10-01 by ngolf