prusa3d/Prusa-Firmware-Buddy #4230: [BUG] MK4S Extruder squeaking with PETG

DataPocalypse42, 2024-10-01:
### Printer model

MK4s

### Firmware version

6.1.3

### Printing from...

PrusaLink

### Describe the bug

After the MK4s upgrade, printing PETG started producing a new squeaking noise.

Here a video I sent to the customer support that was not able to help identify the issue 
https://www.youtube.com/shorts/r_WVYx8WXvM

After multiple extruder rebuilds/cleaning/calibration, idler swapping, I was not able to fix the issue.

But I used the same PETG spool before MK4s upgrade with no issue and I'm only using the classic 0.4 nozzle with STRUCTURAL 0.2 profile.

So I decided to try downgrading to v6.0.4 when everything was working to ensure this was not an assembly issue and by using the mk4 printer profile and the same STRUCTURAL slicer profile, I have only a 4 minutes difference in the 3 hours total printing time and the noise is now completely gone!

I started looking at the diff but was not able to identify which commit can be responsible of this extruder noise.

I will try upgrading version by version and see how to reproduce this issue until the noise will reappear

avehome, 2024-10-06:
Thank you for reporting this. I have the same problem and have also contacted support. The only advice they could give is to tighten or loosen the idler screw. But my screws are exactly as in the manual? I followed there advice anyway but: tighter = more squeaking and looser is slightly less squeaking. But I don't want to loosen it anymore, because then I'm afraid the filament will no longer be gided properly. See also my video and picture of the filament. Looks perfect how the filament is gripped. Also the prints are fine only the sound is annoying...

https://github.com/user-attachments/assets/042ddbe9-e253-4960-9c32-c46f838069f3

[image]
[image]

DataPocalypse42, 2024-10-06:
> The only advice they could give is to tighten or loosen the idler screws. But my screws are exactly as in the manual? I followed there advice anyway but: tighter = more squeaking and looser is slightly less squeaking. 

Thank you for reporting your experience, I just tried on a running print to loosen the idler screws a lot more than previous try and the squeaking noise is gone.
The customer support told me that my screws was ok (they asked for a photo)

Here my screws and filament grip, the running print finished with no issues and no more noise, I hope this will definitely fix!

[image]
[image]

avehome, 2024-10-07:
Wow...that is indeed a lot more inward. I have now done the same, but still need to fine-tune it further. But it's already much better (I still hear it every now and then). Also next time I change the filament I will also check the grip lines. Great that we found a solution, but Prusa really need to change the manual and photo for the default alignment.

DataPocalypse42, 2024-10-07:
I think it's a rare use case because no issue with PLA and assembly manual instructions. On reddit, some people think it's glossy/silk filament surface that can produce this noise and my Jayo PETG have a glossy surface. 

And same here, can still hear it a little bit but much better than my video and acceptable, my head didn't exploded at the end of the day

DataPocalypse42, 2024-12-08:
Hello @avehome, I think I identified the cause of squeaking. Loosening the tension didn't helped and was back after a while.  

This little bit (I don't know how to name this one in english ^^) showed some wearing while servicing the printer, and since I have many idler (one in PETG, ASA and PCCF), I inspect each and they all show wearing :
[image]

So, I tried to cut this bit, tightened as documented by the assembly guide and...... finally the noise disappeared! 

I printed two 5 hours parts and no quality issue or strange noise/clicky thing.

Then I found an idler mod : https://www.printables.com/model/1072020-mk4s-bogie-idler-main-plate

The idler have a different bit but didn't seem to have the same surface contact with the filament but they both seems to not have any critical mechanical usage, so I tried it and it works too.

But don't use the main plate, it's more tight than the original one and is squeaking. 

So I think it confirm that this is the root cause of this annoying noise, I will update if something goes wrong.

( be sure to have an idler backup if you want to try to cut the bit )

avehome, 2024-12-09:
Thanks for the follow up DataPocalypse42. I will give it a try when I do my next maintenance begin next year. I also still have the noise but not always. Prusament PETG Jet Black squeaking more than orange. Will make a backup idler, but first need to order Prusament PC Blend Carbon Fiber with some other parts I need.

DataPocalypse42, 2024-12-26:
Little update after many hours with the bit cut, I can say this eliminated 99.9% of annoyance.
The last 0.1% is related to retractions, I can hear a little one from time to time but compared to my video, it's a drop of water in the ocean. I think there is still a little wearing on the main plate holes, but not too much that can make me crazy.

Hope this finding can help you!

stzoran1, 2025-01-03:
Facing the same issue. Hope Prusa will issue official instructions for solving this problem. It is so annoying. 
https://github.com/user-attachments/assets/078eed35-e414-43e5-b038-9b2c6b4ff869

DataPocalypse42, 2025-01-03:
> Facing the same issue. Hope Prusa will issue official instructions for solving this problem. It is so annoying. https://github.com/user-attachments/assets/078eed35-e414-43e5-b038-9b2c6b4ff869

Can you try the idler "bit cut" as I reported above? just be sure to print a backup part before

stzoran1, 2025-01-04:
Before cutting I tried lubricating this section (I know this is not a solution) just to confirm noise is coming from this part but it didn't help. What I think is creating squeaking noise in my case is friction which is coming from contact between filament and main plate, please check image bellow. 

[image]

Beukie66, 2025-03-11:
Sort of the same problem here when printing PLA.

The slower outer walls make a very annoying I will make a support issue to Prusa.

closed 2025-05-19 by github-actions[bot]