prusa3d/Prusa-Firmware-Buddy #4231: [ENHANCEMENT] Show Image of object to be printed on Display in more situations

Mike-FUT, 2024-10-02:
### Printer model

MK4S / XL

### Describe the enhancement

Currently I can see an image of the object to be printed can be displayed in many situations.
Please consider showing the imaga additionally:
- when a print is waiting in the connect queue and the current one finishes and I remove the objects from the plate, choose "set ready" from the printer screen. Currently it will just start printing without showing an image.
- during a print it would be really nice to be able to see the object on the screen, too, probably by turning the knob or using an alternating screen

Printer606, 2024-10-06:
Yes, this would be nice.

CZDanol, 2024-10-08:
We have some ideas like this as part of BFW-5403

Mike-FUT, 2024-10-09:
> It currently works when setting the time > 30 s and I guess it will work with smaller times when the bug is fixed.

One more update of what I noticed:
The image of the item to be printed never shows with the default setting of 30 s on my MK4s with 6.1.3
But, contrary to what I wrote above, with 32 s set the image didn't show up for several minutes last time I tried a longer print. So there is something rather buggy there...

CZDanol, 2024-10-09:
That part of your issue we're dealing with as a part of #4184