prusa3d/Prusa-Firmware-Buddy #4234: [BUG] Cannot Set Static IP for Ethernet Connection

edtyler, 2024-10-03:
### Printer model

XL

### Firmware version

6.1.3

### Printing from...

None

### Describe the bug

When the Ethernet connection type is set to 'stat', it is not possible to enter the IP address, netmask , broadcast address or DNS server information. 

Note: another bug was entered describing that DHCP does not function. (#4233)

Thus, the Ethernet port is completely disabled on the printer. I cannot use Wifi for security reasons.

### How to reproduce

1) Select 'stat'
2) Try to enter IP address, netmask, etc.
3) Entry fields are inhibited

### Expected behavior

When using a 'stat' connection type, the user must be able to enter the network parameters, such as IP address, Netmask, default route and DNS server IP so that the printer may communicate using Ethernet.

CZDanol, 2024-10-03:
Hello, not a bug per se, just that the feature is not yet implemented :)

For now, you should be able to use the ini file to configure these.

edtyler, 2024-10-03:
Thanks - can you tell me how to edit the .ini file on the printer? Is there a hidden serial port or some other way to access the printer's file system? I have no Ethernet connectivity.

potjak, 2024-10-06:
https://help.prusa3d.com/article/how-to-set-a-static-ip-address_695648

edtyler, 2024-10-13:
Thanks for helping me understand the settings interface. I mounted a USB stick on a Linux box and created the .ini file with these settings:
[eth::ipv4]
type=static
addr=172.16.10.71
mask=255.255.255.0
gw=172.16.10.253

[network]
dns4=10.48.96.37

Next, I used Settings > System > Load Settings from File. The printer loaded the settings, but indicated the connection failed. Both ping of gateway and dns showed 100% packet loss. 

At this point, I started to suspect something else was going on. I brought out my NetTool Pro and observed that the link was going up and down at about a 5 second interval. 

Long story short, the house cable was installed a long time ago and even though it was CAT5, it was not properly pinned for EIA-568A/B. When I configured the link for 10-BaseT, it worked fine. Re terminating both ends of the house cable got the speeds to 1000-BaseT.

Bottom line, the printer works fine, will accept either DHCP or  a statically configured address. This was completely the fault of the house wiring.

Please close this case as USER ERROR. I will also request close of #4233.

CZDanol, 2024-10-14:
Okay, closing :)

closed 2024-10-14 by CZDanol