prusa3d/Prusa-Firmware-Buddy #4235: Tool changer crashes right side and drops tool after cancel

WeaselWasher, 2024-10-04:
### Printer model

XL 2 head

### Firmware version

6.2.0a1

### Upgrades and modifications

Shielded ribbon cable LCD to buddy

### Printing from...

Prusa connect, Ethernet 

### Describe the bug

Unfortunately I was in another room for the actual instance, but I'll describe what I did.

I finished loading new filament, set it to preheat PLA and went upstairs to slice my print and send it to the printer. I sliced a real simple print (vase mode trash can) and sent it via Ethernet connect. (Selected radio button, set printer ready and print)
I then realized that I forgot to scale a dimension to make it more oblong. I cancelled it after 3 min.. then I sliced the new one, and sent it via connect. Set printer ready and print. Connect was saying it was idle. I went ahead and deleted two files from printer via connect.. the one I had previously uploaded and another that I didn't need anymore.
A couple of minutes later I went downstairs to check on the print and found that the tool changer had gone all the way to the right, reached it's extent and unlocked and subsequently dropped the tool.
[image]
[image]

I checked the tool out and then placed it back in it's dock. The screen registered this, with a green checkmark then it flashed a screen I missed and then ended up giving me a USB fault. 
[image]

### How to reproduce

I haven't tried. But I described exactly what I did above. 

### Expected behavior

The tool changer shouldn't ever travel that far to the right with a tool in it.

### Files

https://www.printables.com/model/665632-strong-vase-mode-garbage-bin-can 
Used this STL. Scaled up the z to 32cm and then sliced in vase mode with prusa PLA in tool 1 (everything else default). Then cancelled that, and narrowed it. Made it a little longer, rotated it so it was near max on the diagonal and set it to the printer again.

WeaselWasher, 2024-10-04:
In retrospect, that USB file error could be because the print had crashed and paused when I deleted the file via connect

closed 2024-12-12 by github-actions[bot]

CZDanol, 2024-12-12:
Hello, this looks like #4337, should hopefully be fixed in the next release candidate.