prusa3d/Prusa-Firmware-Buddy #4238: [FEATURE REQUEST]  Child/Wife display and knob lock

HonzaJaros, 2024-10-07:
### Describe the feature

can be long press or in the menu be option to lock display and knob?

wife or kids can mess with it around and gives me headache when they cancel the print

### Expected functionality

o to menu,

lock

long press on the knob or display to unlock, or code?

Printer606, 2024-10-07:
Mabye an option for a password...

Sembazuru, 2024-11-04:
Could also be useful in public Maker Labs to mitigate others from "accidentally" stopping prints.

HonzaJaros, 2024-12-08:
@JakoobCZ hi, any news about this one?

thanks

CZDanol, 2024-12-09:
Hello, we have a similar task in our backlog under the internal ticket BFW-5542 that is currently await consultations with the product department.

HonzaJaros, 2024-12-09:
Great thank you 🙏 Honza Jaros

Kachidoki2807, 2025-07-03:
Hello, how do you think it is possible to "protect" the reset button?
It is a safety feature and I am not sure it is even possible, nor desirable to lock it.

To only prevent accidental interaction with the screen, a cover [like proposed by @spjt](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4705#issuecomment-3014975934) is better than a software solution IMHO.
From there you can imagine multiple levels of "protection", more or less locked, from a simple flip cover to a padlocked one.

developercrocodiles, 2025-10-15:
But this should be implemented as if you press reset then it was not accidental.