prusa3d/Prusa-Firmware-Buddy #4240: [BUG]

Tillquist, 2024-10-07:
### Printer model

XL Multitools

### Firmware version

6.2.0-alpha2

### Printing from...

Prusa Connect

### Describe the bug

I don't quite know how to describe the error, but watch the video and you can see that the print head moves in the opposite direction than it should normally do when starting to print.

Same gcoder runs fine on firmware 6.1.3 without anything.

[Fail with 6.2.0-alpha2 firmware](https://youtu.be/WZuZLFfwz2c)

### How to reproduce

Same error the 2 times I have tried 6.2.0 alpha 2, I experienced the same error on 6.2.0 alpha 1

CZDanol, 2024-10-08:
Hey, to be clear - this is at the beginning of a multitool print?

CZDanol, 2024-10-08:
Internal ticket: BFW-6165

Tillquist, 2024-10-08:
Hi

Yes it is, at the stort up.

Med venlig hilsen

Morten Tillquist Nielsen

Tværstenen 17 (x-apple-data-detectors://0/1)

4281 Gørlev (x-apple-data-detectors://0/1) - DK

CZDanol, 2024-10-14:
@Tillquist I'm failing to reproduce this. Could you please:
* Send me the gcode that's causing this?
* Show me your dock station offsets?

Tillquist, 2024-10-14:
> @Tillquist I'm failing to reproduce this. Could you please:
> 
> * Send me the gcode that's causing this?
> * Show me your dock station offsets?

[image]

[image]

[image]

[image]

Tillquist, 2024-10-14:
Hi

Here is the gcode

---
Bedst Redards

Morten Tillquist Nielsen

Danol skrev den 2024-10-14 16:25:

> @Tillquist [1] I'm failing to reproduce this. Could you please:
> 
> * Send me the gcode that's causing this?
> * Show me your dock station offsets?
> 
> [3].

Links:
------
[1] https://github.com/Tillquist
[2] 
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4240#issuecomment-2411421874
[3] 
https://github.com/

CZDanol, 2024-10-14:
@Tillquist seems like github doesn't support email attachments.

Tillquist, 2024-10-14:
> @Tillquist seems like github doesn't support email attachments.

so how can i share gcode with you?

CZDanol, 2024-10-14:
You can for example upload it somewhere and share the link.

Tillquist, 2024-10-15:
Here it is [Download here](https://wetransfer.com/downloads/6b112f487aeafd8911e2bed2349bdf4e20241015053228/167bd3268440ce4de5a459d133b614de20241015053251/5a2f07?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)

The same file runs perfectly fine on firmware 6.1.3 without problems and it is made in Prusalicer 2.8.1

CZDanol, 2024-10-30:
@Tillquist damn, I thought I have downloaded it, but apparently I forgot. Sorry :( Could you please reupload?

CZDanol, 2024-10-30:
Can I also ask for the exact tool mapping you used?

Tillquist, 2024-10-30:
> Can I also ask for the exact tool mapping you used?

I don't quite understand what you mean?

CZDanol, 2024-10-30:
> > Can I also ask for the exact tool mapping you used?
> 
> I don't quite understand what you mean?

Mapping gcode filaments to physical toolheads. The initial sequence suggests that filament 1 is mapped to tool 4.

Tillquist, 2024-10-30:
[image]

CZDanol, 2024-10-31:
Thanks, and the reupload? Sorry :(

Tillquist, 2024-10-31:
[Download Gcode here](https://www.dropbox.com/scl/fi/1tmge79w8zthw5ynbiuil/Whistle_v2_HCS24-v5_0.4n_0.2mm_PETG-PETG-PETG_XLIS_2h28m.bgcode?rlkey=ywvfwmj38fbukw3vt9wv6tmd4&st=ro5ouii8&dl=0)

CZDanol, 2024-10-31:
Okay, I was unable to reproduce it on my XL on 6.2.0-alpha2 with your file, but I still believe I have found the cause of the crash and I have a different reproducer. It's down in the motor control code for which we have @wavexx who's expert in that, so I will try passing the issue to him.

CZDanol, 2024-11-01:
Okay, we've been hopefully able to fix the issue. Please be so kind and try the next 6.2.0 candidate when it comes out and report here if the issue is resolved for you.

Thank you for your cooperation :)

Tillquist, 2024-11-01:
I do that 👍

closed 2025-01-09 by CZDanol