prusa3d/Prusa-Firmware-Buddy #4241: [FEATURE REQUEST] Retract before mesh bed leveling

slackista, 2024-10-07:
### Printer model

MK4

### Describe the feature

After filament is loaded certain filaments are oozing(including filaments made by Prusa), and accumulating material around the nozzle, causing failed nozzle cleaning. I can reproduce this with high precision. 
[image]

### Expected functionality

 Just add little retraction before mesh bed leveling to prevent oozing.

Racemade, 2024-10-08:
Actually there are workarounds like reducing the temp in the gcode or to buy an mmu:) i think you can add retract per gcode too. But i m with you, this should be released by prusa. Problem occurs especially by petg and the loadcell on the mk4 is not that good as for the xl (i think there is less pressure for the mk4 while cleaning/probing).

Silvenga, 2024-10-09:
Retract wouldn't work, since the bed is normally probed at 170/175 degrees (PLA/PETG). This whole area is controlled by g-code - so PrusaSlicer.

I find that wet filament cases oozing much more often (all my PETG filament doesn't ooze at printing temps when dry).

CZDanol, 2024-10-14:
Internal ticket: BFW-6188

CZDanol, 2024-10-14:
Okay, I've consulted it with our guys:
* We're already retracting a few mm for the nozzle cleaning
* Retracting more wouldn't have much effect, it wouldn't clear the liquid filament from the nozzle, so it wouldn't affect the oozing
* If we wanted to really prevent oozing, we would have to basically do a cold pull (or the procedure we do during the unload sequence), which would increase leveling time and filament usage.

So down the line, rectracting more before MBL wouldn't have an effect.

closed 2024-10-14 by CZDanol