prusa3d/Prusa-Firmware-Buddy #4242: [FEATURE REQUEST] Axis movements and virtual keyboard with smaller increments

svake, 2024-10-08:
### Printer model

XL

### Describe the feature

Sometimes it is necessary for me to move an axis with smaller increments than 1mm, for example when I want to do a manual z-axis calibration or when I just want to check seomething. This is not possible with 1mm increments, and it would be great to have the option to move in smaller increments, like 1/10 and 1/100 like in other firmwares. For this I would also like the point key in the virtual keyboard active as well. In the moment it is greyed out with the current firmware.

### Expected functionality

When in axis movements, I would expect e new window, optimal with graphical buttons for touch control and three buttons for the increments (1mm, 1/10mm, 1/100mm). If you want to know what I mean, have a look at the Repetierhost firmware, where you can move all axes in one screen via graphical interface. Also I never know whether the axis movement is active or not. That leads to axis movement, when I just want to return from the menu and to movements in the menu, when I want to move the axis. With a new window that would be easier, and with virtual buttons on the screen for moving the axis that would be really great.

CZDanol, 2024-10-08:
Hey, we have been considering this under the BFW-5323 ticket, but haven't gotten to it yet.

murk-sy, 2024-11-07:
Having it function somewhat as a key repeat on computers could be especially useful for longer small movements.

For example, if the button is pressed (x indicates movement by selected unit and "tap" sound being played, while - indicates 250 miliseconds of pause)

x----x--x--x--x--x--x--x

Maybe unnecessary depending on how often and in what contexts it's actually used - personally I've never used manual precise movements on a printer, at least as far as I remember.