prusa3d/Prusa-Firmware-Buddy #4244: [BUG] High Fan Speeds can Cause Thermistor Related Panics

Silvenga, 2024-10-09:
### Printer model

MK4S

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

Octoprint/Serial

### Describe the bug

I'm opening this per support live-chat.

It appears that a panic is possible with the MK4 due to increased cooling. When heating above 80 degrees and cooling above 60% (e.g. the default 90% in OrcaSlicer) the heater bed cannot maintain a the temperature, and starts dropping.

For safety, the MK4S panics as the temperature is dropping while the heater is on.

Example of this inability to maintain bed temperature. 

[image]

### How to reproduce

The standard PETG temperature tower generated by OrcaSlicer with a max fan speed of 90% and a bed temperature of 90 degrees.

### Expected behavior

I would expect no panics to occur when operating within the design limits e.g. fan on at 100%. If nothing else, this should be documented (as IMO, any panic should be diagnosed).

CZDanol, 2024-10-14:
Internal ticket: BFW-6186

ivous, 2025-06-02:
Hello @Silvenga, this is HW limit of the printer. Bed temperature is at full power and at the same moment you try to cool it down with fan at full power. I would say the power panic is properly signalled at this moment as there is a risk of loosing adhesion due to low bed temperature. This is simply wrong setting of the slicer parameters. 
The print fan at MK4S is more powerfull compared to previous models, so with older slicer versions this maybe "unexpected". 
We are providing printer profiles and material profiles within PrusaSlicer. From there you will find out that for MK4S PETG the recommended bed temperature is 85C 1st layer and 90C other layers, while fan speed is 20% at first layers and 40% from 5th layer.  With these settings we can guaranteer it si working properly.

closed 2025-06-02 by ivous