prusa3d/Prusa-Firmware-Buddy #4245: [ENHANCEMENT] VFA Improvement

bluerider098, 2024-10-09:
### Printer model

XL

### Describe the enhancement

I have both MK4's and an XL Multitool.  The VFA's are noticably worse on the XL vs. the MK4.  This is particularly noticeable on circular objects. 

I would think the XL with the accelerometer would provide equal results however it does not. 

I'm requesting further tuning be done to the XL to provide better results similar to that of the MK4. 

### Expected functionality

[image]
[image]

bluerider098, 2024-10-10:
> They look like ringing to me. Have you tuned the input shaper frequencies ?

They are set at what the automatic calibration set them at.  I've talked with Prusa Support and they recommended tightening the belts tighter than the belt tuning app recommended, but it makes no visible improvement.  It's much better than before input shaping was introduced, but not near as good as the MK4. 

This is most evident on circular objects.  It's hard to see in the photos, but the smoothness of the surface is far superior on the MK4. 

I'd argue that if additional manual tuning of the input shaping is required, then there is still improvement to be made in the firmware.  I would expect with a Core XY machine to see equal or better print quality of the print than the I3 style machine. 

I'm open for suggestions though.

DoubleStrike, 2024-10-25:
Just curious if you are using 0.4 or 0.6mm nozzles on your XL.

bluerider098, 2024-10-25:
> Just curious if you are using 0.4 or 0.6mm nozzles on your XL.

0.4 nozzles.  

Since my last post I've tightened the belts more and was able to improve the result from the picture above, however my MK4 is still better.  I'm hesitant to tighten the belts further as I'm well above what the tuner app recommends and worry that I may cause other issues.

It feels like there could be improvement in how the input shaper values are calibrated.  I am not an expert by any means, but the movements the print head make during calibration seen very soft and slow compared to what I've seen on other machines.  I don't know that this means anything, just an observation. 

II'll also note that these results are using the structural settings.

andersholck, 2024-12-13:
Shouldn't the XL naturally be worse as it uses the older style 1.8° motors, while the mk4 has the upgraded 0.9° motors.

bluerider098, 2024-12-14:
> Shouldn't the XL be worse as it uses the older style 1.8° motors, while the mk4 has the upgraded 0.9° to combat VFA

That's a good question.  Being the motion system is quite different I can't answer that question.  There's more involved than just the step on the motors. Either way it could use improvement.

closed 2025-02-20 by github-actions[bot]