prusa3d/Prusa-Firmware-Buddy #4247: [BUG] End of filament not recoverable with 0.8mm nozzle

kbob, 2024-10-10:
### Printer model

MK4

### Firmware version

6.1.3+7898

### Upgrades and modifications

0.8mm brass nozzle installed

### Printing from...

PrusaLink

### Describe the bug

When I run out of filament using a 0.8mm nozzle, the printer pushes the filament too far down and can't retract the last bit of filament.  The filament stub remains in the hotend below the drive gears.

### How to reproduce

Print a thing using a 0.8mm nozzle and corresponding slicer settings.  While printing, run out of filament.

I haven't tested other nonstandard nozzle sizes.

### Expected behavior

I expect the printer to stop at the end of the filament and to push the filament stub up 3-4cm so it can be removed, just like it does when I use a 0.4mm nozzle.

### Files

I've included a .ZIP file containing a .bgcode file and two photos.

There is nothing special about the .bgcode file.  It's just a cube with default PrusaSlicer 2.8.1 settings for 0.55mm Draft Mode, Generic PETG, and MK4 IS w/ 0.8mm nozzle.

The first photo shows me cutting the filament going into the extruder after the printer finished the first layer.

The second photo shows the cut end of the filament stuck in the hotend.

[08_nozzle_filament_lost.zip](https://github.com/user-attachments/files/17326838/08_nozzle_filament_lost.zip)

[image]
[image]

CZDanol, 2024-10-10:
Hey, this is happening basically because the printer at least tries to finish the current gcode it's running before it can trigger the filament runout sequence. We are considering a method of how we could trigger the filament change immediately (BFW-4523), but it has a potential to break stuff, so we've been quite hesitant to pursue this further for now.

kengdahl, 2024-10-11:
I have the same issue on a Prusa Mk4S with a .4 mm high flow nozzle.

Tupson444, 2024-10-11:
I guess that because of commands in buffer get executed before pausing, if you are unlucky that the filament runs out on a long extrusion (e.g. a flat wall of a large part) it will be too late to pause after finishing printing those lines. Wide extrusion width and layer height (often used with larger diameter and high flow nozzle) increases chance of this happening because more plastic is extruded per XY distance.

An idea to solve this problem in the slicer: split all long extrusions to multiple shorter commands.

kabelkao, 2024-11-28:
Same bug on MK4S + MMU3. Extruder cant push filament to MMU because filament is too short.

magistersmax, 2025-01-19:
I just got a 0.8mm nozzle on my MK3.5 with Revo and am experiencing the same thing.  Nothing more to add aside from verifying that it's happening to me too and a fix would be hugely appreciated (though it sounds like it won't be a simple one).

closed 2025-03-31 by github-actions[bot]