prusa3d/Prusa-Firmware-Buddy #4248: [BUG] Public github repository seems out of sync

rbuehlma, 2024-10-10:
### Printer model

Does not matter

### Firmware version

6.1.3

### Printing from...

Does not matter

### Describe the bug

The public github repo seems to be out of sync and some commits are missing. If I open the tag v6.1.3 of the current stable release: https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/02104fc75cdb6f0a13b182c3531a6d55c5064dd2 i get the following error:

> This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository.

Could you make sure, the commits are pushed to the public repository since this would simplify contribution.

hlovdal, 2024-10-20:
This is not a repository sync issue, the "warning" is just how github works for commits that are not reachable from any branch. If you run

```shell
git tag commit.02104fc 02104fc
gitk --all &
```

you will see the following

[image]

which shows that the commits that are the behind the 6.1.x releases are only reachable from git *tags* and not any *branches*.

If you have cloned this repository and want to avoid getting that warning just create a branch, e.g. `git branch maintenance-6.1.x v6.1.3`.

rbuehlma, 2024-10-20:
What you just showed, is some kind of security issue of github: https://news.ycombinator.com/item?id=41060102 Not sure if Prusa was aware of this before ;)

Looking at the log of tag v6.1.3 (and master as well), I can find my commit with this hash: https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/72613655afe64e64a164b337dd9d7cab4c396e29

Originally, it was merged with this hash: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3840/commits/3a32d8bb950e5842842d5cf1159af8c2ac7c2ec9
PR: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3840

Comparing these two commits, gives me:

1171 files changed, 59342 insertions(+), 67440 deletions(-)

To me, it looks like Prusa is working on their private fork of the repo and force-pushing the changes from time to time to the public repo making it hard for everyone trying to keep their branch up to date.

bkerler, 2024-10-21:
Main development happens in their private repo (that's how it always has been, sadly). Being open-source (even if delayed) is still better than being fully closed-source imho.

rbuehlma, 2024-10-21:
I agree on that, and developing on a private repo is fine but releases should be built from public available commits. And branches that others rely on should not be rebased and force-pushed.

rbuehlma, 2024-10-28:
I don't understand why this should be the case. You can merge a PR in the public repo and into the private repo as well (or simply merge the main branch of the public repository into your private repositiory). Later when you have a release ready, you could merge your private changes into the public repository. This can all be done without a single force push.

This is the only opensource project I'm aware of that force pushes on the main branch on a regular basis. This should really not be required.

CZDanol, 2024-10-29:
If we did a merge like that, we would have different histories between our public and private repo. We had this some time ago and it was rather a pain to manage.

closed 2025-01-06 by github-actions[bot]