prusa3d/Prusa-Firmware-Buddy #4250: [BUG] API fails to report ATTENTION after heating disabled

rtdog, 2024-10-12:
### Printer model

MK4

### Firmware version

6.1.3

### Printing from...

PrusaConnect

### Describe the bug

The API's /printer, /job, /v1/job all report healthy printer state when printer requires filament change and heating disabled due to inactivity.  One or more of these APIs should report "ATTENTION".

### How to reproduce

1. Start print with a short length of filament, let it run out.  
2. Inspect /printer, /job, and /v1/job APIs and notice /api/printer reports state.text="printing" and state.flags.link_state="ATTENTION", /job reports state="Error", and /v1/job reports state="PAUSED".  This is operating as expected.  
3. Allow hotend to timeout after 30 minutes with "heating disabled due to 30 minutes of inactivity"
4. Inspect /printer, /job, and /v1/job APIs and notice /printer reports state.text="printing" and state.flags.link_state="PRINTING", /job reports state="Printing" and /v1/job reports state="PRINTING".  

### Expected behavior

I expect one or more of the API's to report ATTENTION status instead of PRINTING as the device requires attention and is not printing.

CZDanol, 2024-10-14:
Hey, thanks for the report :) Internal ticket: BFW-6187

bkerler, 2025-08-26:
The root cause is here:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/77255c9c85a25feb22f112724016aba552a1aa3e/src/state/printer_state.cpp#L77-L92

Here's the comment by the devs:
```
Note: We don't consider these attention, so just note the dialog code and slap
it on whatever state we decide, that the printer is in later.
```

The fix should be rather simple by keeping the attention state (returning True) if the nozzle or heaters have a timeout (State: CONNECT_NOZZLE_TIMEOUT or CONNECT_HEATERS_TIMEOUT). @CZDanol can you share an insight why this isn't considered as attention state as stated in the code comment ?

A more complex fix would require that the state is being tracked separately from the ClientFSM, assuming the user isn't "in front" of the printer, but not sure if this doesn't introduce more issues.

bkerler, 2025-09-14:
@3d-gussner looking at 6.4.0-alpha source code, it still returns false on ErrCode::CONNECT_HEATERS_TIMEOUT and doesn't verify its current state